keirf/flashfloppy #78: 1.66MB + unicode

ronnybkk, 2018-03-22:
Hello Keirf,

First, thanks a lot for that piece of software, I'm experimenting with it and having a lot of fun.

As you may know, Gotek drives are popular with Copiers owners: floppy disk based hardware that dumps and load console games, mostly for the SNES/Megadrive consoles.

Most of these copiers support the standard 1.44M IBM PC standard, as well as a non-standard 1.66M (83 tracks / 20 sectors per track). The Super Wildcard DX2 even supports the ED 2.88M format.
The reason for creating this format was to be able to store 12Mbit instead of 11 on a single floppy.
Back then we were using TSR dos programs on PC to read/write on those 1.66M disks.

I'm using a dos tool named bfi to create .IMG files and it works great for 1.44M floppies. However that tool is not able to create anything between 1.44M and 1.68M, so I tried the latter. To my surprise I was able to dump a 12Mbit cartridge and play it back from the Gotek with FF, although loading times were very slow.
On the other hand any operation that consists in creating a floppy image with several files (8Mbit + 4Mbit for example) does not work properly.

So my 1st question: is there an easy way to support this unconventional 83T/20S floppy format?
2nd question : is unicode gonna be supported? I live in Asia where most users would enjoy that feature.

Thanks

Ronny

keirf, 2018-03-22:
1: 83T/20S should already work. Explain more carefully exactly what is not working (read, write, etc).

2: I am happy to support Unicode, but I have little experience of it. The FatFS library seems to have good support (after all the author resides in Japan), but will need someone with an interest to plumb that support into the rest of FlashFloppy. I am happy to advise, or conversely for someone to advise me on how to go about that. This belongs in another ticket, single-issue tickets please: hence I will modify the title of this ticket.

ronnybkk, 2018-03-22:
Thanks for your prompt answer. Since I couldn't see anything pass track 79 on the oled display I assumed it couldn't. Could you tell me how to create a valid 83/20 floppy .IMG file with dd or any other tool so I can return to my tests and get back to you?
Thanks!

keirf, 2018-03-22:
There is no limit to 79 cylinders in the firmware. Certainly the OLED display should run all the way up to T:99. What size is your IMG, exactly (number of bytes)? And do you have an FF.CFG file and if so what does it contain?

Simone2018, 2018-03-22:
Keirf, can you modify the oled to display track till 255? 
you can exclude the ":" if you need more space, like
T:99 -> T100 -> T101

keirf, 2018-03-22:
I could, there's not much point as nothing goes beyond approx cyl 85 I'm pretty sure, apart from HxC direct access mode (hxcmount, hxc autoboot.hfe)?

Simone2018, 2018-03-22:
If is not a great problem, I would be happy.... I made several .HFE floppy with 28 sec/track and up to 255 tracks! 7 Mb floppy in a 720kb drive...

ronnybkk, 2018-03-22:
So regarding the 1.66MB floppy, theoretically it should be 83x20x2x512=1,699,840 bytes but I formatted a floppy on the copier, the total capacity is 1,620,480 bytes. A 12Mbit game on it uses 1,573,376 bytes with 47,104 bytes of free space

My FF.CFG is the one by default, nothing special

ronnybkk, 2018-03-22:
Actually I just formatted a 1.6M floppy with the FDformat PC tool we used back then, it also produces a 1,620,480 bytes disk.
Here are the info provided by the software:
2 Head, 80 tracks, 20 sectors/track, interleave: 1, GAP-length:42, 224 Root Directory entries, 1 sector/cluster, sector-shift: 0:0

Total sectors on disk: 3200
sectors per track: 20
heads: 2
bytes per sector: 512
hidden sectors: 0
boot-sectors: 1
number of FATs: 2
sector per FAT: 10
Total clusters on disk: 3165

Sorry if all that sounds a bit confusing. I believe I was wrong about the 83T/20S, could it be a smaller GAP length that induced a higher capacity on the floppy? I've been poking around with several softwares to try to match the exact specs of that format and I believe this one is most accurate.

What I'd like now is a way actually to create a blank .IMG file with those specs. Any tool or command line on dd I can use to achieve that?

keirf, 2018-03-22:
You should be able to FDformat the 1,699,480-byte IMG file to achieve the above specification. Only difference I think would be the gap (mine is 40 bytes, theirs is 42 bytes, a tiny difference, no real matter I think). And perhaps interleave, depending what they mean by interleave=1.

ronnybkk, 2018-03-22:
Bingo! I succeeded in creating a working image with Winimage custom format, using the above parameters. At first, I used 3165 total sectors, the .IMA file matched the 1.6M floppy size perfectly but FF threw me an error 31. So I changed it to 3200 and it worked.
I need to run more tests as Writing seems to use more space than it's supposed to but at least I can create fully functional 1.6M 12Mbit floppy images.
Also I'll need to find out a proper tool to generate those .IMA files rather than the tedious Winimage.

Thanks again for your help ;)

keirf, 2018-03-22:
3320 total sectors is indeed correct. :) It should not be possible for writes to take more space than they should. Once the filesystem is formatted, physical details like GAP and interleave should not affect logical space use at all. They could maybe affect performance.

closed 2018-03-22 by keirf