keirf/flashfloppy #81: Index unset when track is empty [HFE]

claunia, 2018-03-23:
When a track is not written at all in the HFE file the index is not sent and so the FDC thinks the disk is not there. However, on PC this line is connected to an optodetector that fires when the disk hole passes thru it, or in the case of 3.5", when a magnet in the spindle alters to a detector.

keirf, 2018-03-23:
Can you explain more, like if the HFE is blank?

keirf, 2018-03-23:
HFE images certainly normally fire index pulses. They did as of last I checked, last week. Index pulses normally are only suppressed during seek and immediately after a write (while writing back to USB).

If there is a HFE you think is behaving wrong, please attach it to this ticket so I can take a look.

claunia, 2018-03-23:
I went to HxC and created a singled sided disk, then created a double sided disk, both with 83 tracks, then clicked ERASE SIDE x in edit tools, then formatted both with 22DISK as CP/M SSDD and DSDD 80-track formats respectively, scanned in ANADISK and seen the error when arriving track 80 on the DSDD one, or on the SSDD when ANADISK tries 00.1 and 20.1 to see if the disk is single sided (making it think there is no disk at all). I'll attach the images asap.

keirf, 2018-03-23:
Will be interesting to see. I believe HxC writes 0x11 endless bitcell pattern on erase so there are regular flux transitions (every four bitcells) and FlashFloppy should work with it just fine.

keirf, 2018-03-23:
Send me HFE images before and after formatting with 22DISK.

claunia, 2018-03-24:
http://www.claunia.com/hfeimages.tar.xz

MF1DD.HFE is the blank image created with HxC tools, "supposedly" with only one track.
CPM_BEE3.HFE is it formated as 3.5" SSDD 512x10 with 22DISK.
DSKA0095.HFE is a blank 3.5" DSDD tried to format with 2M that returns "incorrect density" error.
DSKA0113.HFE is a blank 3.5" DSDD tried to format with FDFORMAT ("/F800") that returns disk not ready on track 79. (It starts formatting from end).

keirf, 2018-03-25:
Can't really tell what happened with the 2M disk. FDFORMAT tried to write an extra sector per DD track and ran out of space (reached the index mark before finishing writing the 10th sector). CPM_BEE3 from 22DISK looks fine.

claunia, 2018-03-25:
I attached logic analyze captures to the other issue, also thankfully both 2M and FDFORMAT have the source code available so we can check what are they exactly doing.

keirf, 2018-03-25:
Thanks, I think we need to go after one issue at a time. Perhaps the FDFORMAT one to start with? It's more efficient to deep-dive on one and get that fixed, then move on.

keirf, 2018-03-25:
This one is possibly related to `index-suppression = no`. In any case I'm pretty sure I don't believe it, empty HFEs created by HxC tools certainly have a non-zero bit pattern.

claunia, 2018-03-25:
Attached image and logic analyzer captures at http://www.claunia.com/anadisk.tar

keirf, 2018-03-26:
I'm not sure what anadisk_t79 and anadisk_t80 are traces of (ie how they differ)?

The former definitely has many short pulses on the SIDE signal, which sadly aren't well handled by FlashFloppy as yet, even for HFE images.

Please can you try the following in FF.CFG:
```
side-select-glitch-filter = 50
```

keirf, 2018-03-27:
Updated the title text. Problem seems to be that Anadisk reports "drive not ready" rather than "data not found".

claunia, 2018-03-27:
Note: on normal situation (no data found) it's not an error because it just reports the condition and continues the scan. However with "drive not ready" it stops the current operation, making it impossible to work on single sided disks (even if option to look only one side is chosen it does 00.0, 00.1, 20.0, 20.1, and if .1 doesnt work, stops there).

bleuge, 2018-07-18:
Sydex products were created by Chuck Guzis (22disk, copyqm, teledisk, anadisk, etc) an ubergod of magnetic media. http://www.bitsavers.org/pdf/sydex/
He  is a user at vcfed, and usually writes about old disk stuff.

claunia, 2018-07-18:
It would be great if he opened the source of his tools or at least gave us the file format specifications

bleuge, 2018-07-18:
Sydex was sold to other company years ago, specialized in Foresincs. So, i am sure this is not possible.
You can read the whole stuff googling a bit. If you need some information about some format just ask, maybe I have documentation or source code in my floppy-related archives.

CarlosTex, 2018-12-23:
One of the reasons i asked for Teledisk format support before is because it is capable of dumping some copy protected disks correctly. There would be no need for write support to Teledisk, only to read them. I've had success imaging copy protected disks with Teledisk then convert to HFE then read them with Flashfloppy. Sierra/Taito/Microprose/Broderbund games that used on disk copy protection can be read properly by Flashfloppy. TD0 to HFE works fine, but direct support would be nice.

Anyway, Dave Dunfield, not only created a program that does more or less the same as Teledisk, he also reverse engineered Teledisk Format. So if there is any interest:

http://www.classiccmp.org/dunfield/img/index.htm

http://www.classiccmp.org/dunfield/img/index.htm

keirf, 2019-05-22:
Might as well re-try this one on latest FF v2.x too...

keirf, 2019-06-02:
Please reopen or reissue if this persists on v2.12 or later.

closed 2019-06-02 by keirf