keirf/flashfloppy #83: Formatting in high density

claunia, 2018-03-25:
Formatting in high density works randomly.

keirf, 2018-03-27:
Here's a firmware for you to try. It batches HFE writes so should perform a lot faster on ordinary USB sticks. Let me know how it goes, I can also tune it further if it seems on the right track.

[ff_83_1.zip](https://github.com/keirf/FlashFloppy/files/1852631/ff_83_1.zip)

claunia, 2018-03-27:
it's gotten worse, now it never works, not even on the SSD were all HD formats did work :/

keirf, 2018-03-28:
That's weird, might need to test myself, writing HD images from Supercard Pro perhaps. Might need to speed things up by implementing a buffer cache at least for FAT metadata.

keirf, 2019-05-22:
Please test again with latest FF v2.x.

keirf, 2019-06-02:
Please reopen or reissue if this persists on v2.12 or later.

closed 2019-06-02 by keirf