keirf/flashfloppy #84: FDFORMAT cannot format in any density

claunia, 2018-03-25:
FDFORMAT goes to last cylinder (39, 40, 41, 79, 80 or 81 depending on switches), formats and fails on verify with "drive not ready?" error.

FDFORMAT source code is available in all distribution ZIPs, is in Pascal, and seems to use INT 13h BIOS facilities.

claunia, 2018-03-25:
Logic analyzer capture with both real drive and flashfloppy on http://www.claunia.com/fdformat.tar

keirf, 2018-03-25:
One issue here is that you are running with `index-suppression = no`. Please don't. You can;t undo it by removing it from FF.CFG as it gets remembered in Flash. You must hold both buttons for a few seconds with no USB stick inserted to clear Flash config, or you need `index-suppression = yes` in FF.CFG.

keirf, 2018-03-25:
Now working with `index-suppression = yes`

closed 2018-03-25 by keirf

claunia, 2018-03-25:
Confirmed solved.