keirf/flashfloppy #85: 2M cannot format double density

claunia, 2018-03-25:
2M, and 2MGUI, cannot format double density media.

It tries to format track 0, verifies and then returns "incorrect density" error.

Source code should be available (I have to dig it).

claunia, 2018-03-25:
Logic analyzer capture with both real drive and flashfloppy on http://www.claunia.com/2mfdd.tar

keirf, 2018-03-25:
Same initial issue as #84: Need `index-suppression = yes` (the default).

claunia, 2018-03-25:
Nope, still happening, attached image and logic capture with index supressed at http://www.claunia.com/2mfdd.tar

keirf, 2018-03-26:
Do you have a sigrok trace with a real floppy drive? For comparison, as the behaviour here is weird (initial write for four full revolutions / index intervals, for example).

claunia, 2018-03-26:
Redownload http://www.claunia.com/2mfdd.tar I updated it with a real drive trace.

keirf, 2018-03-26:
Okay similarly messy. Like #81 also worth trying with
```
side-select-glitch-filter = 20
```

claunia, 2018-03-26:
https://github.com/VARCem/VARCem/blob/master/src/floppy/fdd_86f.h this bitcell format, and the corresponding emulators (VARCem and 86Box) support 2M (both DD and HD) formats correctly. Afaik it has been tested with all known DOS formatters. It doesn't support 2MGUI because of timing problems on the emulation side.

keirf, 2019-05-22:
Another one worth trying with latest v2.x. Also if there are issues, you could gather a debug log from the logging firmware in the alt/logfile subfolder (you run your test, eject the disk image, and only then pull the USB stick, then inspect the FFLOG.TXT that was written to it).

keirf, 2019-06-02:
Please reopen or reissue if this persists on v2.12 or later.

closed 2019-06-02 by keirf