keirf/flashfloppy #88: TI-99/4a DSK support

Sinphaltimus, 2018-03-29:
Hello, as per your request, here's mine. :)

There is a need for native DSK (v9t9). Hopeful you can address this in your firmware. 

Here are specifics to the TI's DSK v9t9 formats:
https://www.ninerpedia.org/wiki/Sector_Dump_Format

Here are specifics concerns raised by the active TI community on the format and needs in particular.
Any technical question regarding the DSK v9t9 format can be asked here.
http://atariage.com/forums/topic/276608-important-news-and-i-need-some-help/

Here is a repository of DSK images for testing.
ftp://ftp.whtech.com/Diskettes/

I understand the implementation of this is contingent on other work currently being done on the firmware.

keirf, 2018-03-30:
Thanks, implementation depends on completion of #16 (which appears to be just about done).

keirf, 2018-04-01:
By the way is there any use for PC99 image support? Or is everything available as V9T9 no problem?

keirf, 2018-04-04:
Okay here is an initial implementation, which I have smoke tested.

[ff_88_1.zip](https://github.com/keirf/FlashFloppy/files/1875761/ff_88_1.zip)

It detects track layout/geometry solely by image file size, supporting: SSSD, DSSD, DSDD, DSDD80, DSHD80. An alternative is to detect based on geometry description in sector 0 as described here: http://www.unige.ch/medecine/nouspikel/ti99/disks.htm
The question is whether this can be relied upon for all images (and eg. the format must be extended/modified for HD images, where the allocation bitmap would not wholly fit in sector 0). So some alternatives are:
1. Use file size, fall back to sector 0 for conflicts (eg. DSSD ve SSDD). 
2. Use sector 0, fall back to file size if values in sector 0 are nonsense.
3. ??

Sinphaltimus, 2018-04-04:
As a priority, v0t0 is the preferred dsk format. Adding PC99 support would be a bonus. I'm going to send TI99er folks here shortly to help test and answer any questions.

keirf, 2018-04-04:
Forgot to mention: need v9t9 file suffix or `host = ti99` in FF.CFG to use dsk suffix.

keirf, 2018-04-04:
@mizapf Thank you very much that looks super useful! I think I'm pretty close to your code on the filesize hinting, and I can take a look at your use of the volume info block :)

keirf, 2018-04-05:
Here's an updated firmware handling more formats: SSDD, DSSD80, and the 16-sector-per-track DD formats.

[ff_88_2.zip](https://github.com/keirf/FlashFloppy/files/1880786/ff_88_2.zip)

keirf, 2018-04-16:
Closing this pending user testing. Please open new tickets for any issues raised by user testing.

closed 2018-04-16 by keirf