keirf/flashfloppy #89: Not working with the Acorn Archimedes A440/1 running RISC OS 3.10 (ADFS)

simoninns, 2018-03-30:
I've installed a Gotek flashed with the latest v0.9.16a from the releases page.  When loaded a .adf file (800K) the initial load of the directory works fine (and the disc contents are shown) however, attempting to load anything further from the disc results in a 'drive empty' error (or 'disc not present' - depending on what directory entry I try to load).

Here is my FF.CFG:

host = acorn
index-suppression = no
twobutton-action = rotary
display-probe-ms = 6000
display-type = oled-128x32
display-off-secs = 120
display-on-activity = yes
step-volume = 18

My jumper settings are JC and S0 closed.  The gotek is the only floppy present (directly replaced the original floppy).

keirf, 2018-03-30:
Try adding `interface = shugart` to your FF.CFG.

simoninns, 2018-03-30:
Total magical wizardry :)  I will test some more, but that seems to have cured the issue.

keirf, 2018-03-30:
Might also be worth trying `index-suppression = yes`. I don't know whether the other setting is required across the whole Archimedes family.

simoninns, 2018-03-30:
The A440/1 is one of the earlier ARM machines (and is known to be a bit picky about it's floppy drives).  I've done some basic testing (loading games) with the firmware and everything looks good so far.  Tomorrow I will stress it a bit more with formatting and the like (and also try the index-suppression - although I think = yes will work fine).  If I find anything more, I'll report it as an issue, but I think you can consider this one closed now. 

Thanks for the prompt help :)

simoninns, 2018-03-31:
Tested both 640K and 800K .adl and .adf images; reading, writing, formatting and verifying all works.  The following settings were used:

`host = acorn
index-suppression = yes
interface = shugart
`

keirf, 2018-03-31:
Thanks I will update wiki and then close this ticket.

closed 2018-04-01 by keirf