keirf/flashfloppy #90: With index mode display shows F-F

tamis-laan, 2018-04-02:
So i'm not sure if there is something wrong or i'm doing something wrong. I have flashed my Gotek and created a Fat32 usb with FF.CFG which has the "nav-mode = indexed" configuration set. It also contains DSKA0000.hfe up to DSKA0003.hfe image files. However when I insert the usb into the Gotek the display only says F-F, and i'm unable to select floppy disks. Am I missing something or did flashing process go wrong somehow?

keirf, 2018-04-02:
If you remove FF.CFG and reset the Flashed config (hold both buttons for a few seconds with no USB stick inserted) and then insert the FF.CFG-less USB stick -- does it work?

Displaying "F-F" means USB stick not recognised, usually.

tamis-laan, 2018-04-02:
Nope the problem persists, could this be a device issue? The Gotek is connected to a Yamah Disklavier self playing piano.

keirf, 2018-04-02:
No it is a USB stick issue. Please try another or reformat that one. Put nothing on it for now but a single HFE image.

tamis-laan, 2018-04-02:
I took another USB stick and it worked! Maybe it has something to do with the fact I used osx to format the usb stick to fat32?

keirf, 2018-04-02:
Yes probably macos has done something like put on a GPT rather than legacy MBR and partition table. There will be some way of doing it the right way on macos but it may not be the default!

closed 2018-04-02 by keirf