keirf/flashfloppy #91: Acorn ADFS 'S' format not supported

simoninns, 2018-04-04:
There is no support in flash floppy for the Acorn ADFS 'S' format which is 40T SSD 160K; attempting to use a 160K .adf image results in *ERR*31*

This format was supported by the Acorn BBC Master 128 ADFS and (to be honest) was barely (if ever) used - but, if possible, should be included for completion.

keirf, 2018-04-04:
That should be very easy. I will take a look.

keirf, 2018-04-05:
Please test the following firmware:

[ff_91_1.zip](https://github.com/keirf/FlashFloppy/files/1879473/ff_91_1.zip)

simoninns, 2018-04-05:
Tested by formatting, verifying, writing files, reading files and creating directories.  Works perfectly  :)

keirf, 2018-04-05:
Thanks, will be supported in the next release (v0.9.17a).

closed 2018-04-05 by keirf