keirf/flashfloppy #93: Problem with OLED Display - works only after the second switch on.

sirdavid2, 2018-04-05:
I have recently connecded OLED Display to my Gotek, as described in FlashFloppy Wiki, but there is a problem with it. At first I thought it didn't work and probably my display was faulty. But it turned out that I have to switch off the computer for one or two seconds, switch it on again and then the display works. It never works after the first switch on and always works after the second one. Do I need pull up resistors or something?

keirf, 2018-04-05:
You should add a 4.7k resistor to R3 on the display PCB. Or you can work around in firmware by adding `display-probe-ms = 3000` to FF.CFG configuration file (https://github.com/keirf/FlashFloppy/wiki/FF.CFG-Configuration-File).

Let me know if that value (3000) works and if so I will increase the default value (which is currently 2000) before closing this ticket.

sirdavid2, 2018-04-05:
3000 works!

Squirrel61, 2018-04-05:
On my Gotek, the display only turns on when a USB stick is present. Without USB stick, it remains black. Is that related (meaning I can make it turn on by playing with the probe time), or is it supposed to stay black when no storage is available?

keirf, 2018-04-05:
@Squirrel61 display may time out after a while of inactivity and be reactivated by stick insertion? But it should certainly illuminate at power on in any case (at least with default configuration / no special values in FF.CFG).

commit 3310fd8f (2018-04-05): FF.CFG: Increase default display-probe-ms to 3000
Fixes #93
closed 2018-04-05 by keirf