keirf/flashfloppy #94: Gotek creates load errors when connected externally

DarkKnight94, 2018-04-06:
I recently flashed a gotek drive using the v0.9.16a firmware and the instructions provided by the author. I mounted my gotek as an internal drive and tested, everything is working fine this way.

The issue presents itself when i try to use my gotek externally. My setup consists of an Amiga 500 with kickstart 1.3, a gotek unit flashed with FlashFloppy  v0.9.16a, a boot selector and a cable to connect the Gotek externally.

Everytime my gotek is connected externally to the Amiga, my disk drive refuses to load Xcopy or Workbench and presents checksum error 880. Games however seem to work correctly. If i disconnect my Gotek from the Amiga,  Xcopy or Workbench are loading without any issues. I have tested with and without the boot selector and experience the same symptoms.

For testing purposes, i flashed my Gotek with the Cortex firmware and it worked correctly... If you need any help recreating the issue, let me know. I am more than happy to assist

keirf, 2018-04-06:
Hi there,

I think you have edited this a few times to narrow things down? First of all, you said earlier that this was tested with Cortex firmware with no issues -- but you no longer mention Cortex so did you discount this observation? If Cortex worked reliably in the *exact* same setup, same Gotek, etc then it becomes significantly more likely this is a firmware error rather than a setup error.

Is the A500 vanilla (no accelerator)? What interface cable or board do you use to connect the Gotek as DF1? What jumpers are set at the rear of the Gotek?

Finally, describe just one step-by-step scenario to reproduce this error on your setup. Attach or provide for download the disk images that are involved. This will allow me to attempt reproduction myself if we cannot pin this on a problem in your setup.

I won't be around to test this myself or implement fixes until w/c 16th April, but I will be online to advise on this ticket.

DarkKnight94, 2018-04-09:
Hello again!

As you correctly mentioned, i made a few edits as i ran more tests and ruled out some factors. For example, i was able to reproduce the same issue with the Cortex firmware, hence i don't mention it anymore.

To help you recreate my issue, i have concluded to 3 different scenaria:

1) I mount R-Type or R-Type II on the gotek (df1)and try to boot workbench or X-copy from physical floppy(df0)-> checksum error 880

2) I disconnect the gotek from my amiga and try to boot workbench or X-Copy from floppy-> they boot correctly

3)I mount bubble bobble to the gotek(df1) and try to boot workbench or X-Copy from floppy(df0)-> they boot correctly

If loaded from the gotek directly (df0), R-Type and R-Type II load with no issues. At this point i tend to believe the issue is adf related, but maybe it can be fixed by firmware?

Thanks for your time

DarkKnight94, 2018-04-09:
Since i am out of town, i will upload the files and the URL to the cable in a few days, sorry to keep you waiting!

My gotek only has one jumper located on the s0 position. My amiga only runs an A501 trapdoor expansion and has no accelerator or other enchancements. My board is a rev. 5, even though i don't think it matters

keirf, 2018-04-09:
How are you connecting the Gotek as DF1? What is the interfacing cable or board that you are using, specifically? The problem can be there too.

DarkKnight94, 2018-04-10:
First of all, i am using the following cable to connect the Gotek to the Amiga:
https://www.ebay.com/itm/External-Amiga-drive-adapter-cable-for-Gotek-HxC-FF-Usb-emulator-A500-boot-hack/272363110859?hash=item3f6a1b41cb:g:6Z4AAOSwYDZac39N
It does not feature any controllers, but the Gotek itself loads the games correctly, so i suppose it should not be the problem. The jumper for my Gotek always remains on df0

As requested, i am including the 3 ADF files i used so you can also test them yourself, if you have some free time and wish to do so. For your convenience, i labeled them as "incompatible" and "compatible" to clarify which ones create the issue.
[Roms.zip](https://github.com/keirf/FlashFloppy/files/1892450/Roms.zip)

keirf, 2018-04-10:
I have the same cable. I will give it a try perhaps next week.

keirf, 2018-04-10:
By the way, could the problem simply be a checksum validation error on *DF1*? What happens for example if you boot XCopy or Workbench with nothing (or 'compatible' ADF) in DF1, then switch to an 'incompatible' one after boot? I think what you may be seeing is simply the expected behaviour of AmigaDOS when it finds no filesystem on an inserted disk. Bear in mind an ADF always has valid AmigaDOS **sector** layout, but this does not automatically imply it has an AmigaDOS **filesystem** on it too. I would wager that Bubble Bobble does have a valid filesystem (or enough of one to keep AmigaDOS happy) while R-Type does not...

DarkKnight94, 2018-04-11:
Hello again

It could be possible, since upon closer inspection i noticed the gotek LED flash once before the error comes up.

If i change to an 'incompatible' image after i boot the utility, the same error comes up again. A colleague of mine with knowledge on Amigas suggested that this might be result of a game with no bootable dos sector