keirf/flashfloppy #95: SDU file format

Lantizia, 2018-04-08:
Hey,

This is basically a feature request... I could try to supply a patch, but it's such a simple change and it'd likely take me waaaaay longer than someone else being unfamiliar with the language/code, but I'll try at some point if needs be.

Basically I have an AST computer (remember those?) and they provided floppy disk image files on their recovery CDs... this was so you could recreate the floppies if you wished to install everything manually (e.g. DOS, Windows, MS Works, etc...)... rather than just re-imagining the whole machine from the CD.

These image files end in extension .SDU and basically AST used the software from this guy to make them (so it's likely more than just AST users that have SDU files)...

http://www.saberman.com/SABDU/index.htm

On that site is a zip of his main program for making SDU files and includes some example SDU files.  It also has a .DLL for making your own end-user friendly extraction program (like on my recovery CD from my AST).  Here is the one from my CD (I've only included MS Video for Windows SDU files as at least that's freeware)... [astdu.zip](https://github.com/keirf/FlashFloppy/files/1887190/astdu.zip)

The short version of this story is... please support SDU files...

**All you have to do is ignore the first 46 bytes and then just treat them like a regular IMG file.**

p.s. I haven't actually got a floppy emulator drive yet... can someone point me to a good company/person to buy one in black with an OLED already built in?  Preferably someone who is contributing back to this project in some way?

keirf, 2018-04-16:
Not one I'm at all aware of, previously. If they are just IMG with a header then you can strip that yourself, on a machine with the `dd` command-line utility (eg Linux, MacOS):
```
dd if=my.dsu of=my.img bs=1 skip=46
```

Lantizia, 2018-04-16:
Yeah I'm aware I can do that, I already do exactly that now.

But it'd be nice not to need to?  Admittedly it's a convenience thing, but it looks good to add another file format - if nothing else?

I am vaguely interested to see if anyone else finds this post who uses/used SABDU too.

Lantizia, 2018-04-16:
Oh wow, quick turn around :)

Although my Gotek has technically arrived (with FlashFloppy already flashed)... it's this one with the cool black rotary selector (which I've not seen before - but is presumably supported in the default FlashFloppy firmware)... https://www.ebay.co.uk/itm/222905730623

I've gutted the components already so that they can go into another non-modified Gotek I've got coming in black (from Hong Kong, so a slow boat that's not arrived yet)...

Will get back to you.

keirf, 2018-04-16:
Please reopen this ticket if issues arise in future testing.

keirf, 2018-04-17:
I think SDU files must have zeros truncated from end of image file, which makes geometry detection (eg. double density vs high density) based on file size impossible.

I imagine the SDU header contains geometry info. Or else perhaps it only supports HD 1.44MB images? Do you have any other SDU images, ones which might not be HD 1.44MB?

Lantizia, 2018-04-17:
Well here is all the ones I have (the sizes of the files do vary)
[image]

The first 46 bytes of them are all the same as each other (so the contents of those 46 bytes in all files will be the same as the 46 bytes you've got in the sample files I put in the astdu.zip above - as this full list of files below includes vfwr11d1.sdu and vfwr11d2.sdu).
[image]

I believe all the images were intended for HD 1.44MB floppy disks - I've never known it to be used for anything else.  The ASTDU program (their rebranded SABDU for customers) even shows this image which implies it too :)
[image]

The SAD DU website (which based on its age - it's a miracle it is still up) doesn't seem to say one way or the other which kinds of disks it was meant to be used with... I'd probably need to use his software (the one that can author SDU files) to find it, will try that in a few days.

Lantizia, 2018-04-17:
p.s. My last post has been heavily edited as it was wrong in maaany places, posting this to make sure if anyone has read the old one - re-read :D

keirf, 2018-04-17:
I have emailed the author to ask about the SDU header format. For now, FlashFloppy v0.9.17a is hardcoded to 1.44MB PC geometry.

Lantizia, 2018-04-17:
Brill - that comes up T:00.0 for both vfwr11d1.sdu and vfwr11d2.sdu now :)

Even mswareba.sdu and qkse3uk2.sdu work too, despite them having a different header for some reason (but presumably still the same length).

keirf, 2018-04-17:
They may come up T:00.0 okay but not actually work. Looking at their header they have value 9 where most SDU images have 18. This would indicate a 9 sectors-per-track DD image rather than 18-spt HD. Actually the triplet (80, 2, 9) which could be (cyls, sides, spt) appears twice in the header -- I'm not sure why! I hope the SABDU author will help :)

Lantizia, 2018-04-17:
I mailed him a few days back... just a one liner basically with a link to this page - in case he found it funny people were talking about this format/program again after many years.  This was before we got more in depth.

I did notice that sabdu@saberman.com bounces but sab@saberman.com didn't - so maybe there is someone still behind that address :D

keirf, 2018-04-17:
I think for now, given that all images seen so far have the same (C,H,S) geometry triplet repeated in their header, I will just pick one arbitrarily and use that to set up the geometry in FlashFloppy.

keirf, 2018-04-17:
Okay, please try this firmware, which auto-detects geometry based on SDU image header and probably works better for images mswareba.sdu and qkse3uk2.sdu:

[ff_95_2.zip](https://github.com/keirf/FlashFloppy/files/1919547/ff_95_2.zip)

keirf, 2018-04-18:
I got a helpful email response from the SABDU author. I will revise the handling in FlashFloppy as necessary tomorrow and then close this ticket again after posting an update test firmware.

Lantizia, 2018-04-18:
Wow that's great, any chance you can post what he said?

keirf, 2018-04-19:
He tells me the last update of SAB DU was April 1st, 1995 :)

He also sent me some source code which has let me clean up the sdu handler a tiny bit (the code was already correct though, so not attaching another firmware here, just test the previous one, ff_95_2).

closed 2018-04-19 by keirf