keirf/flashfloppy #96: E31 error on Atari stf with old german roms

ikonsgr74, 2018-04-11:
Hi,

 Atari st with old german roms doesn't seem to cooparate well with flashfloppy. 
It gives E31 error on lcd screen:
http://i.ebayimg.com/images/g/riAAAOSw7btazj3g/s-l1600.jpg
http://i.ebayimg.com/images/g/3WYAAOSwMk1azj3X/s-l1600.jpg
  It happened with 3 differant atari st so far, all with old german roms installed, so i guess there might be an issue with this.

keirf, 2018-04-11:
E31 means FlashFloppy does not support the selected image file. It is not affected by the host that the Gotek is connected to.

So what do you have on the USB stick?

keirf, 2018-04-11:
Probably will work if converted to HFE. Certainly STX is not directly supported by any Gotek firmware.

closed 2018-04-11 by keirf