keirf/flashfloppy #98: E-MU Emulator II - HFE Images Write/Format issues.

il-Sandro, 2018-04-17:
Hi all there.

I'm opening this issue after several test done with Keir.

first of all, In order to avoid misunderstandings, 
- E-MU refers to E-Mu Emulator II Sampler.
- Emulator refers to actual Gotek Floppy Emulator.
 
Got a GOTEK with Flash Floppy 0.9.16a / Oled Display.
It reads flawless .HFE images i have, but not write them as it should.

When engaged, the machine starts to write data, but after a while (each time on different blocks) it stops and main E-Mu display says "Read Error".
- tried with several Usb Sticks, nope.
- writing thru block/sectors seems very clumsy with a kinda random pace.

**** Upgraded to 0.9.16z RC ****
Same stuff.
- tried with several Usb Sticks, nope.
- writing thru block/sectors seems way better with steady pace.
- Tried with different configs from FF (Interface: Shugart /IBMPC - Track Change - Index suppression) nope.

**** Tried to step back to 0.9.6a. **** 
Writing is much slower and clumsy and still gaves errors during writing most of the times, so i think it's not a regression from latest revision.

keirf, 2018-04-17:
Can you attach a pair of HFE images here, before and after being written to by FlashFloppy? It might be interesting to see if FlashFloppy has screwed up a write, or whether it was just too slow and Emu timed out on a read following a write.

I believe the Emu support is probably all there in FlashFloppy, it's just showing up subpar HFE write performance.

keirf, 2018-04-17:
Also, here is another firmware to try, with tweaked HFE write path:
[ff_98_1.zip](https://github.com/keirf/FlashFloppy/files/1920600/ff_98_1.zip)

Still please do send over the before/after HFE images though, you can do those with the stock 0.9.17a firmware.

il-Sandro, 2018-04-17:
Installed Update ff_98_1

Seems way worse in terms that issue spots very early by stopping write routine within first 30 blocks.
Here a written image with errors.

[Empty_Disk_080.hfe.zip](https://github.com/keirf/FlashFloppy/files/1921975/Empty_Disk_080.hfe.zip)

keirf, 2018-04-18:
Thanks, is the HFE from ff_98_1 or 0.9.17?

il-Sandro, 2018-04-18:
Image Empty_Disk_080.hfe has been written by ff_98_1.

keirf, 2018-04-19:
Nothing too bad in the HFE. It looks like EII asserts write gate before it starts producing write data, so there's a sequence of 00 bytes at the start of the write area. That shouldn't really be a problem though. Nonetheless here is a quick firmware to test which drops those 00 bytes. Send me an HFE written by this firmware if it fails:

[ff_98_2.zip](https://github.com/keirf/FlashFloppy/files/1927088/ff_98_2.zip)

il-Sandro, 2018-04-19:
Hi, tried FF_98_2
At least i was able to format an HFE Image with emulator II.

Here attached the formatted HFE with embedded Emulator 3.0 OS ( first 10 blocks occupied).

[FF_EII_3_0_frmtd.hfe.zip](https://github.com/keirf/FlashFloppy/files/1929679/FF_EII_3_0_frmtd.hfe.zip)

il-Sandro, 2018-04-19:
Here the disk written with errors with FF_98_2

[Empty_Disk_081.hfe.zip](https://github.com/keirf/FlashFloppy/files/1929695/Empty_Disk_081.hfe.zip)

keirf, 2018-04-23:
There doesn't seem to be anything especially wrong with the above HFE images. I don;t really know why there would have been a failure in above Empty_Disk_081 -- is it "Read Error" again?

Does FF_98_2 seem more reliable than stock release v0.9.18a? It is overall a bit surprising, really.

il-Sandro, 2018-04-23:
Hi Keir,
i'm at work at the moment, but i can tell you that the same issue happens
on another Gotek upgraded E-MU Emulator II, so this excludes an issue on my
machine.

This evening i could make more test about.

Thank you.
Best regards,
Sandro.

keirf, 2018-04-23:
That's useful to know as it does largely discount EmuII host issues. However the issue was always likely to be in FlashFloppy. I believe HxC gained Emu2 support with the help of someone with technical expertise in the Emu2 community. Since Emu2 has a custom disk hardware interface and software driving it, there will in all likelihood be some expectation of a real-floppy-drive performance characteristic which FlashFloppy is currently missing. The issue is working out what that is, and what the correct tradeoff is to achieve the required level of conformity in that respect, possibly in exchange for less accuracy in something that Emu2 doesn't care about.

il-Sandro, 2018-04-23:
Well yes the issue is always "Read error".

Kyroflux blog/Forum seems the best resource for data and tips about it:

https://forum.kryoflux.com/viewtopic.php?t=898
https://www.youtube.com/watch?v=DrQvV2uFlU8
https://forum.kryoflux.com/viewtopic.php?t=38
https://xa.yimg.com/df/emulatorII-list/EII+Diagnostics+Manual.pdf?token=O4wz_0YOYcriKcuynAg6abrDCdZZAzJGXxXbi2S8tfK4SFaOwqkqkl4kPhva2yKoB08Vun7Z2lZLyNF6L3AixT5vFOhgSSGEHN1Lg0JLj0_4LdCzro8jjhIYoEtquhNPdqYI9z0KKA&type=download

keirf, 2018-04-24:
Was ff_98_2 any better than stock firmware? It sounded like maybe it was, but it could be random chance, depending how long you tested it for.

il-Sandro, 2018-04-24:
Now a trial and error test on both Firmwares:
Here are test made with 98_1 installed.
[98_1.zip](https://github.com/keirf/FlashFloppy/files/1941970/98_1.zip)

- Empty Disk 000: Formatted with 98_1 OK

- Empty Disk 001: Formatted with 98_1 OK
Then tried to write a copy of an original floppy on it . FAILURE on track n°11.0  E-mu Emulator II says: “Drive not ready” ( First 10 Block are occupied by E-Mu OS 3.0).
Tried again to write on it, error on track n° 20.1 E-mu Emulator says “ Disk read error” 

- Empty Disk 002: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 11.0. E-Mu Emulator Says “ Disk read error”

- Empty Disk 003: Clean untouched HFE image from HxC website. 
Tried to write on it, FAILURE on sector 24.1 E-Mu Emulator Says “ Disk read error”.

- Empty Disk 004: Clean untouched HFE image from HxC website. Tried to write on it, FAILURE on sector 11.1 E-Mu Emulator Says “ Disk read error”.

- Empty Disk 005: Clean untouched HFE image from HxC website. Tried to write on it, FAILURE on sector 29.1 E-Mu Emulator Says “ Disk read error”.

- Empty Disk 006: Clean untouched HFE image from HxC website. 
Tried to write on it, FAILURE on sector 15.1 E-Mu Emulator Says “ Disk read error”.
Formatted with 98_1 OK
Tried to write again on it, FAILURE on sector 15.1 E-Mu Emulator Says “ Disk read error”.

il-Sandro, 2018-04-24:
Here are test made with 98_2 installed.
[98_2.zip](https://github.com/keirf/FlashFloppy/files/1942093/98_2.zip)

•	Empty Disk 010: Formatted with 98_2 
FAILURE: Drive not ready on sector n°67.1
•	Empty Disk 011: Formatted with 98_2 OK

•	Empty Disk 012: Formatted with 98_2 OK
Then tried to write a copy of an original floppy on it .
FAILURE on track n°31.1 E-mu Emulator says “ Disk read error”

•	Empty Disk 013: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 17.1.
E-Mu Emulator Says “ Disk read error”

•	Empty Disk 014: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 11.1.
E-Mu Emulator Says “ Disk read error”

•	Empty Disk 014: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 22.1.
E-Mu Emulator Says “ Disk read error”

•	Empty Disk 015: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 22.1.
E-Mu Emulator Says “ Disk read error”

•	Empty Disk 016: Clean untouched HFE image from HxC website.
Tried to write on it, FAILURE on sector 11.1.
E-Mu Emulator Says “ Disk read error”
Tried several times to write it but it always stucks to sector 11.1 with same error message.

keirf, 2018-04-24:
Fair to say there's not much between 98_1 and 98_2, right? How about stock v0.9.18a?

il-Sandro, 2018-04-24:
Here are test made with v0.9.18a installed.
[0.9.18a.zip](https://github.com/keirf/FlashFloppy/files/1942249/0.9.18a.zip)

• Empty Disk 020: Formatted with v0.9.18a : 
FAILURE DURING FORMATTING: Formatted the 80 Tracks, but when writing embedded OS on HFE, it failed during track 01.1 write.

• Empty Disk 021: Formatted with v0.9.18a  
FAILURE DURING FORMATTING: Error during track 52.1 write.
E-mu Emulator Says “Drive not ready”

• Empty Disk 022: Formatted with v0.9.18a  OK

• Empty Disk 023: Clean untouched HFE image from HxC website
Tried to write on it, FAILURE on sector 24.1.
E-Mu Emulator Says “ Disk read error”

• Empty Disk 024: Formatted with v0.9.18a  OK
Tried to write on it, FAILURE on sector. 11.1
E-Mu Emulator Says “ Disk read error”

• Empty Disk 025: Clean untouched HFE image from HxC website
Tried to write on it, FAILURE on sector 28.1.
E-Mu Emulator Says “ Disk read error”

• Empty Disk 026: Clean untouched HFE image from HxC website
Tried to write on it, FAILURE on sector 14.1
E-Mu Emulator Says “ Disk read error”

il-Sandro, 2018-04-24:
_Another possibility: would direct .EII image support be useful? Could you work with EII sector images instead of HFE images okay (eg. are all your disks available in that format or convertible to it)?_

Well, maybe i've misunderstood something.
i mean, if i place .EII images on the USB stick and then into FF i just can't see them during browsing, i see just an empty folder with no images into.
How could i use them if i can't see them in image directories?

Running v0.9.18a at the moment.

keirf, 2018-04-24:
FF doesn't currently support EII images, but I could make it do so, if you can work with them instead of HFE.

il-Sandro, 2018-04-24:
_FF doesn't currently support EII images, but I could make it do so, if you can work with them instead of HFE._

It would be a great improvement, no need to convert them into HFE anymore, which is a quite annoying batch process for large directories.

keirf, 2018-04-24:
Ok I need to understand the format a bit more first but shouldn't be that tricky. Could you point me at some EII images and corresponding converted HFE images? Maybe zip some up and attach to the ticket.

Another advantage of EII support is that writes back to USB would be smaller/quicker, so may satisfy EII host timings better.

il-Sandro, 2018-04-24:
Original E-Mu Emulator II Library in .EII format ( emuiios.emuiifd is present into folder prepared for HFE conversion with HxC software)
[VOICE.zip](https://github.com/keirf/FlashFloppy/files/1942412/VOICE.zip)

Same Library converted in HFE using HxC Software with already embedded Emu OS ( emuiios.emuiifd )
[VOICE.zip](https://github.com/keirf/FlashFloppy/files/1942426/VOICE.zip)

keirf, 2018-04-24:
Thanks those will be helpful :)

keirf, 2018-04-25:
Oh I see.. so emuiios.emuiifd must get embedded into every physical disk image. So to do this on FlashFloppy would require two images open -- emuiios and the data. That's a bit different...

keirf, 2018-04-25:
Regarding the HFE images, there is something different about each track that raises the read error. I can see it visually. I will need to dig into the track contents some more to see if I can work out what's up.

il-Sandro, 2018-05-21:
Any progress about this?

keirf, 2018-06-13:
By the way, could you test latest firmware using HFE images and with the following in FF.CFG:
`index-suppression = no`

il-Sandro, 2018-06-13:
[FFtest180613.zip](https://github.com/keirf/FlashFloppy/files/2099710/FFtest180613.zip)
Nope. Same issue.

keirf, 2018-07-10:
I've looked at the HFE images a bit more. The bad tracks are a bit odd, it's hard to make out what's going on. It could yet be FF being too slow somehow. Debug logs would be useful. Is there any chance you could collect serial logs from a debug build? This would involve connecting up a programming serial dongle while the drive is connected to the EII, and collecting logs via terminal software on the attached PC.

il-Sandro, 2018-07-10:
I have an Emuser which could connect the machine to a PC.
http://users.skynet.be/emxp/EMuSer.htm

Could be useful?

keirf, 2018-07-10:
Probably not useful, unless it can obtain diagnostics/logs of some kind from the machine. The aim with the serial dongle is to get diagnostics from the Gotek drive. Basically to collect more data on what is going on here, as I don't have a good theory at the moment.

il-Sandro, 2018-07-10:
There's no kind of support about .EII image?

Otherwise, the only thing i have is a USB to TTL dongle.
Nothing more than this, i'm sorry.,

keirf, 2018-07-10:
A USB/TTL dongle is exactly what you'd need. Did you program the Gotek via the dongle? You would connect it in the same way.

Another thing to try is the attached HFE image, consider it blank, and format it on the E-mu machine and see if it works any better. It is half the size of the HxC supplied image and may work better if performance is the problem.

[emu_halfrate.zip](https://github.com/keirf/FlashFloppy/files/2180706/emu_halfrate.zip)

il-Sandro, 2018-07-10:
Did you program the Gotek via the dongle?
*Yes.  *

* Another thing to try is the attached HFE image, consider it blank, and
format it on the E-mu machine and see if it works any better. *
*Format on Hfe Images works way better than writing file.*
*Anyway, I'll try and keep you posted.*

il-Sandro, 2018-07-10:
Is there a particular software to record and export logfiles?

2018-07-10 16:21 GMT+02:00 Sandro Codazzi <sandro.codazzi@gmail.com>:

keirf, 2018-07-10:
That emu_halfrate image, I know format works fairly well anyway, but all I mean is to format it on the machine and then try doing some ordinary writes to it after. And see how that goes.

Software for recording log files... any serial terminal program should have facility for displaying received characters and then saving them all into a log file. I use 'miniterm.py' myself which is a very dumb terminal program, shipped with the Pyserial python package.

il-Sandro, 2018-07-10:
I have both MAC OSX and Windows environments but i never do that stuff, any
advice/software usage would be useful to be more precise.

Thanks :)

keirf, 2018-07-10:
For example perhaps Coolterm on Mac.

il-Sandro, 2018-07-10:
Well a little recap.

I should keep the Gotek connected to the Emulator, Then i have to:

   - Connect the TTL's  TX RX terminals as in programming configuration;
   - The Gotek's two connectors of programming shoul be connected via
   Jumper ( YES/NO)
   - Since the Gotek's power alreadyconnected, the +5V and GND terminals of
   TTL interface should be left disconnected (YES/NO)

keirf, 2018-07-10:
Connect Tx rx and ground but leave 5v unconnected. Alternatively you can connect the 5v wire too, but then you must disconnect the normal floppy drive power lead.

il-Sandro, 2018-07-10:
Only one question.
To be programmed, these two pins must be connected by a jumper ( see pics)
Is this connection required or must i leave the two pins disconnected?

​

keirf, 2018-07-10:
You do not need the programming jumper. It puts the Gotek in programming mode which you do not want.

Also remember you need me to send you a debug build. You will not get any serial output from a normal FlashFloppy release.

keirf, 2018-07-11:
Never mind on that HFE image then. Perhaps I messed up creating it somehow.

Regarding the serial logs remember you need a debug build! I will attach one here.

keirf, 2018-07-11:
Here's a debug build, producing log output at 115200 baud. That's quite slow but should be supported by any terminal software:
[ff_98_3_115200.zip](https://github.com/keirf/FlashFloppy/files/2182982/ff_98_3_115200.zip)

il-Sandro, 2018-07-13:
Hi Keir,

busy days here.
I'll make all checks after August 1th.

Thank you.
Best regards,
Sandro

keirf, 2018-07-15:
It's fine, I will be on holiday 28/7-11/8 so will investigate further in mid August.

Invincing, 2018-10-18:
This sounds very similar to the problems I've had with my Akai S1100, as described in issue https://github.com/keirf/FlashFloppy/issues/173

Try setting "pin02 = nc" in FF.CFG to see if it helps.

thestudioman, 2018-12-12:
Hi Keirf
i Also have an Emulator II with the Gotek i have the Problem Described above, i've tried the latest Firmware "Flash Floppy v0.14" i even tried "pin02 = nc" in FF.CFG that Invincing Sugested still not working! not able to Format and Write  or copy to Gotek Reading from Gotek and copying to 5.25 Floppy Works
Help would Be Appreciated
Michel

thestudioman, 2018-12-12:
Hi Keirf
i Also have an Emulator II with the Gotek i have the Problem Described above, i've tried the latest Firmware "Flash Floppy v0.14" i even tried "pin02 = nc" in FF.CFG that Invincing Sugested still not working! not able to Format and Write or copy to Gotek.
Reading from Gotek and copying to 5.25 Floppy Works
Help would Be Appreciated
Miche

thestudioman, 2018-12-13:
Hi Keirf
i Also have an Emulator II with the Gotek i have the Problem Described above, i've tried the latest Firmware "Flash Floppy v0.14" i even tried "pin02 = nc" in FF.CFG that Invincing Sugested still not working! not able to Format and Write or copy to Gotek.
Reading from Gotek and copying to 5.25 Floppy Works
Help would Be Appreciated
Michel

il-Sandro, 2018-12-21:
Hi Thestudioman.

After months of struggling i noticed that the only .HFE image that works properly on write routine is "Empty disk 000".
I can write on it hundred times with no error.

If i try to save same data on anther empty disc i have errors.

So since i prefer to play music than found what's the matter with that i save data on that disc  then rename the image and then load again another " Empty disk 000" into the usb key.

This is the only workaround found that works for my Emulator II.

Hope this helps.
Cheers.

thestudioman, 2018-12-21:
Thanks but i think more Research would Help!
By The Way i Found the Perfect ff.cfg for Ensoniq EPS
here it is with the v2.4.9 OS and "PocketDemoSong.img"
it works perfectly i can loads all the Pocket Demo Song in One Shot Including Midi
let Me Know ifr You Find something Else For Emulator II
Michel

On 2018-12-20, at 7:13 PM, il-Sandro wrote:

Hi Thestudioman.

After months of struggling i noticed that the only .HFE image that works properly on write routine is "Empty disk 000".
I can write on it hundred times with no error.

If i try to save same data on anther empty disc i have errors.

So since i prefer to play music than found what's the matter with that i save data on that disc then rename the image and then load again another " Empty disk 000" into the usb key.

This is the only workaround found that works for my Emulator II.

Hope this helps.
Cheers.

�

keirf, 2019-05-22:
I hear this is now working well as of v2.10a.

closed 2019-05-22 by keirf

raybellis, 2019-07-22:
I need help with the jumper settings for this, please!

I'm using the original E-mu straight through cable (no twist) with an edge-connector-to-34-pin adapter.   I want to boot from the GoTek, and use the original FDD as drive 2.    At the moment the GoTek appears to be completely unrecognised by the system and it's still booting from the FDD.

On the remaining FDD I've got the drive selection jumper in the second position.   I've no jumpers at all on the GoTek, and have the interface type set to Shugart in the `ff.cfg` file.

raybellis, 2019-07-22:
I do now have it working, for reads at least.

The original FDD (still) has the resistor pack removed, and the drive jumper in position 2.

I've put a jumper across S0 on the Gotek.

That allows me to boot from the Gotek, and read files from both disks.

il-Sandro, 2019-07-22:
Hi Ray.

Here the Jumper positions of original FDD:
[image]
[image]
[image]

One of the two original drives should have a resistor array inserted on RP1 chip socket.
Remove it if you want to use the FDD as Drive 1.

and the FF.CFG

[FF.CFG.zip](https://github.com/keirf/FlashFloppy/files/3419133/FF.CFG.zip)

Hope this could help.
Best,

chickeneps, 2020-02-06:
Did this ever get resolved? I would like to put a Gotek with Flash Floppy on my EII, since the floppy drives seem to be failing. A step-by-step would be helpful.

keirf, 2020-02-06:
I have had successful reports on FlashFloppy in the past year, so yeah I think it's fixed. You should be able to install a Gotek with S0 jumper and suitable HFE images and away you go... Noone has published a precise step-by-step afaik but it didn't sound super tricky, if you can get a boot HFE image which I suppose is needed?

chickeneps, 2020-02-06:
I think the HFE image and the S0 jumper should be easy - would you suggest the latest FF version? But my main qusiton really is - above Ray and Sandro are talking about jumper settings on the EII itself, and removing a resistor pak? Or was that addressing dealing with older FF versions?

keirf, 2020-02-06:
I think that's dealing with keeping an original drive working?

[3 more replies]