keirf/flashfloppy #99: No success on Amiga w/ adf

ojaksch, 2018-04-17:
Build a Gotek w/ speaker and OLED which is working fine in an Atari ST and PC, but no luck with an Amiga 1200 (set as internal/first drive for all comps). I can hear the typical "tock---tock" (looking for disk), the OLED is showing the mounted image and jumps from track 2 to track 3 and back, but Workbench isn't showing the DF0.

Jumper is set to S0
FF.CFG: interface = shugart
Software 0.9.16a and 0.9.17a tried

Any ideas? I ran out of mine...

keirf, 2018-04-17:
It will be an Escom A1200 with mod to use a PC drive. Try FF.CFG: interface = ibmpc

If that works, you can optionally undo the Escom mod on your board, so that `interface = shugart` works, and gives better compatibility with game and demo track loaders which often expect the Shugart RDY signal.

See here for how to undo the mod: https://www.youtube.com/watch?v=G6fYOjTYvXM

I will add this explanation to the Wiki.

ojaksch, 2018-04-17:
Jeeze, you're right! Setting FF.CFG to `interface = ibmpc` works - and this damn wie IS present on my board. Will remove it tomorrow and report.
Many thanks for your work and help!