keirf/flashfloppy #100: Long delay on scrolling through SSD/DSD

Yesideez, 2018-04-18:
I've got about 300 files on my 1GB USB stick, carefully selected files from the SSD games collection from TOSEC, #, A-C.

When I scroll through with my rotary encoder, there's an almost half a second delay between it switching from one to the other. If I turn my dial quickly, about 20 turns, it only changes by about 4 disks.

I've also noticed that SSD or DSD appears between the disk number and track number - can this instead be put on the end of the name as it only appears briefly anyway.

keirf, 2018-04-18:
Does it scroll quicker if you have fewer images per folder. Say less than 100?

Yesideez, 2018-04-18:
No, just the same. I've moved all my files into a Games folder as I've added a Beebug folder and it contains 111 SSD files and it's running at the same speed.

keirf, 2018-04-18:
How about if you have very few files, say fewer than 10? Or what about if you try another USB stick?

Also, does the rotary encoder change selection every click albeit slowly, or does it require multiple clicks to change selection?

Yesideez, 2018-04-18:
Looks like it's fixed itself. Not sure why but the DT and CLK wires the wrong way round to make the dial work the opposite way are now the right way and guess what - it scrolls fast :)

The only issue I now have is when scrolling, each disk title doesn't start at the first character which makes it very difficult to see where I am in the list. I have to pause every now and then and see where I am.

keirf, 2018-04-18:
The idea was to allow you to scroll along to disk number part of a TOSEC name and more easily select between multi disk titles. Not sure how well it works and yours isn't the first complaint. Might change the behaviour next release and see if anyone complains.

closed 2018-04-18 by keirf