keirf/flashfloppy #101: Add some way to write-unprotect image

SukkoPera, 2018-04-19:
A feature that I would find very useful (on Amiga at least) would be some way of remove (or better toggle!) the write protection on a floppy image.

This would be done on a floppy during image selection, before the floppy is actually inserted.

If the floppy is the one currently inserted, then simulate eject + unprotect + reinsert.

I'm not sure what button this could be mapped to, but maybe if the encoder is installed, the up/down buttons could be assigned different functions, and this could be one. The other one could be creating a blank floppy or duplicating the current image as mentioned in #71.

keirf, 2018-04-20:
Already possible. Mount the image then hold eject button (that is button 3) for two seconds to toggle write-protect status. Then click that button one more time to re-mount the image.

If you have only two buttons then press both buttons instead of button 3 above. You will need `twobutton-action = eject` in FF.CFG.

I'm not sure this toggle behaviour is documented yet. I'll add it to the wiki.

SukkoPera, 2018-04-20:
Oh, great! I don't think this is mentioned anywhere in the Wiki. Will test tonight, closing the issue anyway.

closed 2018-04-20 by SukkoPera