keirf/flashfloppy #102: Different versions of the 0.91 128x32 OLED Display

RetroClinic, 2018-04-22:
I've noticed that there are different versions of the 0.91 OLED display being sent out from the China sellers. One type has no identifying markings, and has R5 and C9 either side of a transistor Q1 near the connector. This type works fine, and comes on within a second of power being applied.

Another version is made by "DIY MORE", and doesn't have those extra components next to the transistor, now called U2. This display sometimes doesn't power up at all, and can be made to by power cycling two or three times. When it does come on, it takes 2-3 seconds to power up.

Is this a slight difference in the communications protocol? I've tried with the config set to auto, and to force it to the OLED display, but neither setting makes any difference.

keirf, 2018-04-22:
Same protocol but they have a resistor in the reset circuit which is too large and causes them to take ages to power on, and FlashFloppy can timeout meanwhile. In v0.9.18a I increased the timeout to three seconds, you can try a larger value by for example `display-probe-ms = 5000` in FF.CFG. Or you can do a hardware mod to properly fix the display to have a sensible power-on time, described here (a few posts down): https://torlus.com/floppy/forum/viewtopic.php?t=3240&start=60

Let me know if your diymore displays seem to need a larger display-probe-ms value than the current default (3000).

RetroClinic, 2018-04-22:
Yep, that's sorted it. I've not modded the OLEDs as it seems a bit confusing as to what value to use, and what will shorten the life of the display. I've just increased the value to 5000 as suggested, and they all power up now.

Thanks for the swift support!

closed 2018-04-22 by RetroClinic

Squirrel61, 2018-04-23:
@keirf : That's great! It seems I also have one of the affected displays so I will check the resistors and swap them.
@RetroClinic : according to the findings on the forum, R3 and R4 have been swapped on certain diymore displays. R3 should be 4k7 and R4 should be 560k. R3 can be corrected by soldering a 4k7 parallel to the existing 560k resistor in one of the ways indicated on the forum. The 560k will not noticably influence the resulting value and the reset time will be around 50ms, like expected. R4 is a bigger problem, keeping it at 4k7 seems to shorten the life of the display. To correct this, the 4k7 resistor needs to be removed and replaced by a 560k one. This might not be as hard as it sounds, because SMT resistors can easily be removed by heating one end of the resistor for some time while keeping sideways pressure on it. After 2-3 secs, the resistor will be hot enough to melt the solder on the other end and it will move away from the solder pads. After that, it should be possible to resolder the right resistor, being an SMT or normal (preferrably 1/8W for size) version. Any other option will shorten the life of the display, although I don't have any idea about how much the damage will be. Maybe it will still keep running for years with the wrong resistor value.

RetroClinic, 2018-04-23:
Thanks for that. I'll try that mod on one of the displays I have, see what difference it makes.

RetroClinic, 2018-04-23:
Just tried it and bingo! Works a treat, it boots up almost straight away now, thanks! Chinese quality control at it's finest!