keirf/flashfloppy #103: Error: lba change test failed ! write issue ?

Jeroven, 2018-04-23:
Hello, I own an Amiga 500 Rev 6 with 1mb ram and I recently bought a basic Gotek USB floppy drive emulator coming with fw FF 0.9.16a and file selector v1.65a. I also bought an external cable and a df0/df1 switch selector to boot from the gotek without removing the original floppy. Since I experienced sound and graphic issues with some games, I updated the fw to 0.9.18a and things get a lot better. However, after a couple of days of list editing with the integrated file selector (and sporadic hanging on save, which forced me to reset) , the error message above appeared. 
I tried USB stick formatting, USB stick changing and rewriting fw to no avail. Pressing in the A500 power connector - it was a couple mm slipped out - seemed to solve the issue, but after a while the error message came out again. 
The drive memory slots are still usable and I can modify them with the windows app HxC Edit. 
I did so many tries that I'm somewhat confused: the issue can really be power related or some change in the fw can be decisive? 
Thanks.

keirf, 2018-04-23:
Try downgrading to 0.9.16 again and see if that still works for you. 0.9.16 is still available for download and downgrade is same process as upgrade.

Jeroven, 2018-04-23:
Ok, the fw downgrade doesn't make any change, but I just run your systest and I've got some ram errors:
https://drive.google.com/file/d/1h7mqQg05W6kMp0djdZVOoEsxlxwQhDj7/view?usp=drivesdk](url)
Can you tell me if the bad ram is the one on board or the expanded one? Shouldn't I get more issues in loading adfs (most of them run fine)? Can the Lba change test issue be related?

keirf, 2018-04-23:
The error is in the RAM expansion. It may not affect every location in the RAM expansion so it may be down to luck whether it bites on any specific disk. The Lba change test issue is probably related. Just try pulling the RAM expansion out and see if things get better.

Jeroven, 2018-04-23:
It really seems to be the ram expansion to cause the LBA issue, tough the first time after I removed it I still got the error...I will test it for a couple of days more, in the meantime I will probably order another expansion card, and then I'll close the issue...thank you again!

keirf, 2018-04-24:
You should indeed strip the system back, and test with Gotek as DF0. Perhaps run SysTest in that configuration too, and run the memory test for say half hour and see if that runs reliably on the base memory.

closed 2018-04-24 by Jeroven