keirf/flashfloppy #106: 720 kB support?

raybellis, 2018-05-02:
This may just be a documentation issue, but I can't see anything that indicates whether this firmware would allow a HD device such as the SFR1M44-U100K to also emulate a lower density 720 kB floppy, as used for example in the Roland S-550 sampler.

keirf, 2018-05-02:
It is supported, the specific model of Gotek does not matter as they only differ in firmware, which gets replaced with FlashFloppy anyway.

I will add a wiki doc.

keirf, 2018-05-02:
This is now done, in new wiki page Gotek-Compatibility.

closed 2018-05-02 by keirf

keirf, 2018-05-02:
I've clarified it a bit. I don't want to bang on about 720k and 1.44MB too much as a hasty read makes it sound like those are FF's primary focus.

raybellis, 2018-05-02:
Thanks Keir - that's perfect.

The reason this came up is that there was a major discussion today on a FB group for Roland Samplers about which drive emulator to buy, with some asserting that the S-550 (and others in that series) requires a 720kB drive, with the S-750 needing a 1.44MB drive.

Whilst true with the stock firmware, my suspicion (hence this question) was that with FlashFloppy it probably didn't matter and that it would support both on any model (and other formats too).

FWIW, it seems that the SFR1M44-U100K is somewhat cheaper than the 720kB version, even though it's just the firmware that's different.