keirf/flashfloppy #107: Support for Gotek's SELECTOR.ADF

SukkoPera, 2018-05-03:
Any particular reason why this is not supported in lieu of HxC's AUTOBOOT.HFE? If it's just the lack of docs I think I can help, since I have reverse-engineered the format some time ago.

keirf, 2018-05-03:
Is it worth supporting? It's less good isn't it?

SukkoPera, 2018-05-03:
I don't know, I have never used HxC's firmware.

I think it might be good to support it for those who are used to it and want to switch to FF for the write support, plenty of formats or any other reason.

Squirrel61, 2018-05-03:
The title says “Gotek” but I think you mean “Cortex”?

SukkoPera, 2018-05-03:
Well, whatever ;).

keirf, 2018-05-03:
I haven't had any complaints about HxC selector vs Cortex's, in fact it's generally accepted that HxC one is better. So there's no compelling reason to do this work.

SukkoPera, 2018-05-03:
I see, thanks anyway.

closed 2018-05-03 by SukkoPera

v-retro, 2018-05-03:
I would agree supporting the Cortex SELECTOR.ADF wouldn't add value.  The
HxC selector is a better implementation, showing more information.