keirf/flashfloppy #109: Support UKNC and DVK dsk format

xolod79, 2018-05-09:
UKNC and DVK this is the Soviet cloned PDP-11 system with custom floppy controllers (MZ: and MY: disk drivers of RT-11 OS).
There is an standard DSK image format for these floppy disks, which is a byte copy of 512 bytes per sector, 10 sectors per track, 80 tracks, two sides. 
But it does not work directly with DSK on FlashFloppy. This is associated with a known problem in the ROM program of the disk controller. It consists in the impossibility of reading 0 sector on each track (blocks 0, 10, 20 and so far). since the formatting of these sector is not standard.
The difference from the standard is the lack of IAM, GAP1 and SYNC after GAP1.

keirf, 2018-05-09:
Could you attach a working HFE image here please?

xolod79, 2018-05-09:
[http://www.kpxx.ru/%D0%A3%D0%9A-%D0%9D%D0%A6/%D0%A1%D0%BE%D1%84%D1%82/SystemDisks/RT-11%20v05.07/%D0%A3%D0%9A-%D0%9D%D0%A6%20001.hfe](url)

xolod79, 2018-05-09:
[УК-НЦ 001.zip](https://github.com/keirf/FlashFloppy/files/1988094/-.001.zip)

keirf, 2018-05-09:
Try this firmware. You need `host = uknc` in your FF.CFG file.

[ff_109_1.zip](https://github.com/keirf/FlashFloppy/files/1989020/ff_109_1.zip)

keirf, 2018-05-09:
Try the following firmware. You will need `host = uknc` in your FF.CFG file.

[ff_109_2.zip](https://github.com/keirf/FlashFloppy/files/1989036/ff_109_2.zip)

See the patch here (specifically the changes to image/img.c): https://github.com/keirf/FlashFloppy/commit/c62cb4d59bbc7c8d9b3b0dda4b5d738d57d9f0f3

xolod79, 2018-05-10:
thanks, very quick solution. Firmware running OS RT-11 is loaded, but sometimes there are read errors. Trying to gather more information why this is related.

keirf, 2018-05-11:
I'm not accurate enough in representing the format yet. I will have another go...

keirf, 2018-05-11:
Try this (`host = uknc` still required):

[ff_109_3.zip](https://github.com/keirf/FlashFloppy/files/1994278/ff_109_3.zip)

xolod79, 2018-05-12:
Checked: formatting, reading, writing everything works.
Excellent work. Thank you.
We will check on different systems and environments.

commit 4e1f7b92 (2018-05-14): img: Support "host = uknc", 10*512 track with no index postamble
and a weird extra sync after every crc. For UKNC and DVK systems.

Fixes #109
closed 2018-05-14 by keirf