keirf/flashfloppy #110: Support for weak sectors for .DSK (eDSK)

TomDDG, 2018-05-11:
Apologies if this has been raised before or is already possible (couldn't see anything). 

Some old copy protection routines relied on weak sectors which would return random contents if an attempt was made to read them whereas a copy of the disk would return the same contents each time and therefore fail the check. More details can be found [here](http://simonowen.com/samdisk/plus3/) with the following extracted from Simon's page which may be what is required in the FlashFloppy emulator?

> "For this protection to work in emulators they must support the EDSK extension to use multiple copies of error sectors, ideally cycling through the different data copies each time the sector is read. Fuse, SpecEmu and Spin support this, though other emulators may vary the final byte in the sector in order to pass the weak data check."

keirf, 2018-05-14:
Yes, happy to take a look. Point me at some example images. I think someone also by email gave me advice on this some time ago, so I can dig that out too.

TomDDG, 2018-05-14:
I've put a couple of floppy images with the weak sector copy protection into the following [zip](http://tomdalby.com/speccy/Test.zip). I ripped these using the latest version of [SamDisk](http://simonowen.com/samdisk/) and have added the output from the rip so you can see the CRC error sector read.

Hope these help.

keirf, 2018-07-06:
Please try the attached firmware. It seems to work okay for me on a couple of random re-dumped games I tried (Afterburner and Continental Circus, both on Spectrum +3):

[ff_110_1.zip](https://github.com/keirf/FlashFloppy/files/2169352/ff_110_1.zip)

TomDDG, 2018-07-06:
Excellent job. I've just run through the re-dumps I've posted and 9 out of the 14 now work fine, although Robocop loads it crashes after the speech which is weird and might be my copy. Into the Eagles nest crashes the GoTeK but that is bad copy I think! The even better news is I tried a handful of the TOSEC dumps which didn't work before and they also now work so re-dumping isn't going to be needed for all disks. Anyway my initial findings below:

Test Drive 2 - both re-dump and TOSEC version now work
Turbo Outrun - both re-dump and TOSEC version now work
Robocop - re-dump works, TOSEC still doesn't (this is a weak sector protected disk so re-dump was needed)
Footballer of the Year 2 - both don't work
Ghouls N Ghosts - both don't work
Heroes Barbarian 2 - both re-dump and TOSEC version now work
Double Dragon 2 - both don't work
Back to the Future Part 2 - re-dump works, TOSEC still doesn't 
The New Zealand Story - both re-dump and TOSEC version now work
Rolling Thunder - both don't work
Summer Games 2 - both don't work
Outrun - re-dump works, TOSEC still doesn't (this is a weak sector protected disk so re-dump was needed)
Afterburner - re-dump works, TOSEC still doesn't (this is a weak sector protected disk so re-dump was needed)
Continental Circus - both re-dump and TOSEC version now work

and a few TOSEC disks I randomly tested now work, meaning no re-dump necessary:

Chase HQ 2 - TOSEC works
Dan Dare 3 - TOSEC works
Dynasty Wars - TOSEC works
Final Fight - TOSEC works
North and South - TOSEC works
Robocop 2 - TOSEC works

keirf, 2018-07-06:
All work for me except Rolling Thunder and Summer Games 2, which happen to both use Alkatraz protection (none of the others use Alkatraz). I don't know why the other titles aren't working for you, as they work quite reliably on my +3 now.

As for Alkatraz, I'm not sure what's up. There's no weak-sector data in those EDSK images, just a weird track with 18 256-byte sectors. It should be an easy one. Maybe someone like Simon Owen could advise on it?

keirf, 2018-07-06:
Closing this now as broadly the work for weak data is done. Most titles work. The remaining problem images need addressing one by one (or at least protection scheme by protection scheme). We can open new tickets if need be.

closed 2018-07-06 by keirf

keirf, 2018-07-10:
Here is a test firmware claiming 100% compatibility with +3 images in TOSEC:

[ff_tosec_done.zip](https://github.com/keirf/FlashFloppy/files/2179427/ff_tosec_done.zip)