keirf/flashfloppy #111: Memotech -  Wite and Format Errors

stevendt99, 2018-05-13:
Fully described here

http://www.primrosebank.net/flashfloppy/showthread.php?tid=25&pid=39#pid39

keirf, 2018-05-14:
[cut and paste]

I am getting errors when writing and formatting media in a FlashFloppy Gotek when using my Memotech FDX system.

Reads appear to be working fine.

The first symptoms were apparently random errors when copying files from other media (real floppy disks or hfe images in a HxC SD card device). The problem seemed to be when copying multiple files, the copy would start OK, then sometimes, not very frequently, I would get an error writing to the FlashFloppy image (IMG file). The source files were known to be OK and the FF image was believed to be good).

The problem appears to be much worse when I try to format media with FlashFloppy.

I spent quite a while trying various images, different USB sticks, even different computers and different Goteks without being able to put my finger on the exact problem. I was initially using USB3 compatible flash drives but the symptoms also occur with older flash drives too. I have tried low-level formatting the flash drives to no avail.

The problems also occur with hfe images as well as IMG files.

To try and get a better understanding of the nature of the problem, I have done some tests comparing the behaviour of the FF Gotek with my Type C HxC SD Card device.

With an hfe image loaded in the HxC, I can format the image on the system without error. The format takes around 02:16. I can repeat this without issue (10 times in a row is fine).

With the same image in the FF drive, formatting fails at least 50% of the time.

When it does work, the format takes around 01:16, i.e., significantly faster. 

The increased speed would be great, but I am wondering whether FF is going too fast for the disk controller to keep up?

Is there any way to throttle the USB write speed?

Would the config "side-select-glitch-filter" help, if so, what would be a good value to try?

Are there any other config options that might be appropriate to change?

Issue is present with all Firmware versions, including 18a and 19a.

keirf, 2018-05-14:
Can you please attach an HFE formatted by HxC on the Memotech? It would be good to check how the layout compares with that generated by FF for IMG files.

The speed diff on format could be various things, it's not necessarily harmful. But can't discount that of course. It is nothing to do with USB write speed I'm quite sure, nor is 'side-select-glitch-filter' going to help.  You could try `track-change = realtime` but that's a bit of a punt and, if it doesn't help, be sure to explicitly change it back to `track-change = instant` (else the non-default value gets cached in Flash!).

Are you able to run a debug build of FlashFloppy, and obtain logs via serial dongle (the same kind used for serial method of firmware programming)?

stevendt99, 2018-05-14:
I have created a couple of new hfe files using the HxC Floppy Emulator PC software. The two files attached have different interleave values (1 and 3), but the correct one is interleave 3. I just wanted to see if there was any difference with the 1 value (there wasn't) and have just attached it for completeness. The one that you should look at is the I3 file. 
Note, I am not using the IMG format at the moment, I am using hfe just to try and get a common basis for comparison between HxC and FF.
The attached files format without error on the HxC but suffer frequent failures with the FF flashed Gotek.
I use the USB method for flashing, so have not used the debug connector in anger, though I do have a USB to TTL serial device, so should be able to run a debug version of the software if you create one.
Regards
Dave
[HxC.zip](https://github.com/keirf/FlashFloppy/files/2000748/HxC.zip)

stevendt99, 2018-05-14:
Didn't mean to close this just yet

stevendt99, 2018-05-14:
OK, I will try the side-select-glitch-filter change.

I first noticed the issue with very infrequent, but troubling, write errors with IMG files. I think that I have also seen the problem with writes to hfe too, but I have been doing most of my testing with formatting as that seems to be where I can see the problem more often which makes the test results more certain than trying to detect the less common write errors.

keirf, 2018-05-15:
I remember what I should have asked for --- please attach HFE image after an unsuccessful format and, if you can, a separate one for an unsuccessful write. This is the best chance to get clues at what's wrong without getting debug logs.

stevendt99, 2018-05-15:
The attached zip contains 4 hfe files, cloned (on a PC) from a single file that had formatted successfully on the FF Gotek., then copied back to the Gotek.
Image A - not modified further on the Gotek
Image B - format failure on the Gotek, Gotek indicator showing 03.0
Image D - files copied OK from HxC to Gotek with no errors
Image C - the same set of files copied repeatedly until I got a failure
The files copied successfully 4 times and failed on the first file (COLDBOOT.COM) the 5th time, the Gotek reported track 04.0.
After each successful copy, I erased the files under CP/M. As you probably know, CP/M just writes E5 to the first byte of each directory entry, the underlying data is still there, so you might not be able to see much from this failure.
You should see the directory entries starting at Track 0, Side 1, Sector 11
[Archive.zip](https://github.com/keirf/FlashFloppy/files/2005097/Archive.zip)

keirf, 2018-05-15:
Thanks I will find some time to take a look at these. Hopefully something will jump out.

stevendt99, 2018-05-15:
Thanks - good luck! - FYI, the COLDBOOT.$$$ file in the disk with the bad copy error is the filename that CP/M creates when a new file is opened. It gets renamed to the original (.COM) when it is closed after the file has been written successfully

keirf, 2018-05-15:
There doesn't seem to be any physical corruption resulting from the errors in B and C, however all images have corruption on track 51.0 which I guess comes from the original format on FF which apparently succeeded! This may be a separate issue than the "spurious" write/format failures. I wonder if Memotech gets upset at FlashFloppy "going silent" while writing back to USB.

It might be worth trying `index-suppression = no` in FF.CFG (again, explicitly set back to the default value if it doesn't help).

stevendt99, 2018-05-15:
I have just tried index-suppression = no, the image formatted the first time (that error on 51.0 was not there this time, but I did not check other areas of the image) but the format failed the next time. 

When the format fails, the software displays that "an error condition has occurred in the service routine", I'm not sure what that means, but if the controller is somehow checking the disk for a good format write (readback of the sector perhaps?), then it does work most of the time.

keirf, 2018-05-15:
Okay, I think `index-suppression = no` could very much help. Memotech FDX generates disk-ready from regular index pulses. It expects index pulses every 360ms minimum. So long seeks or long writebacks could violate this assumpion. Had the same problem on some Acorn systems (BBC and Archimedes) and developed `index-suppression = no` to solve it there.

Please try with that option, with IMG files.

keirf, 2018-05-15:
Please avoid HFE for now, that 51.0 error clearly shows there's another issue at play here which IMG should not experience. Interested to know if formatting IMG can be reliable with the index-suppression cfg option.

stevendt99, 2018-05-15:
OK, will try that now, just to reiterate though, it did not improve the issue with the hfe file.

keirf, 2018-05-15:
Understood, but I am much less confident in HFE reliability than IMG reliability. A tonne of users are regularly using the latter; the former, not so much.

stevendt99, 2018-05-15:
Yes, I drew the schematics - not that I fully understood them :-)
They were based on some really crappy scans from the '80s that I tried to tidy up t help me try and understand the disk controller's workings - not that that was 100% successful :-)
Just tried to format an IMG, with the same error unfortunately with index-suppression = no

stevendt99, 2018-05-15:
Oh, just went and had a look at my website. The drawing that is on there that I redrew in KiCad is for the SDX disk controller. I am currently using the FDX disk system, and although I have an FDX disk controller schematic in progress, it isn't finished yet. I imagine that the index generation is the same, but I will have a look at the FDX controller now.

keirf, 2018-05-15:
Oh yes, worth a look :) Perhaps that was a red herring in that case.

stevendt99, 2018-05-15:
The FDX disk controller is quite a bit different to the SDX one, that reminds me that I do need to spend some time getting that schematic updated, but there is quite a bit of work to do. 

The scans on the website for the "SM1 Floppy Disk Controller" are the most current at the moment. The SM1/FDX was a much bigger controller - it supported 4x8" or 4x5.25" drives on different interfaces, but only the 5.25" disks matter now. (If you look at the schematic, you will see that for some unfathomable reason, Memotech reversed the pin numbers on the Floppy Interface cable.)

keirf, 2018-05-15:
Very different but still has a '123 for generating READY from /INDEX if the floppy header does not have a READY pin. So still potentially `index-suppression = no` does matter. What USB stick are you using for these tests?

stevendt99, 2018-05-15:
OK, so even if it doesn't fix it at the moment, going forward, index-suppression = no should be set for Memotech devices?

I have tried other USB sticks (not with index-suppression = no though), but am currently using a Sandisk USB 3.0 64GB stick. It has been low-level formatted with FAT32 to try and make sure that there were not any bad cells, but I don't think there would have been , it was a pretty new stick.

keirf, 2018-05-15:
I would strongly consider that option, though of course we've seen no improvement at all so far... Perhaps try another stick at random now?

keirf, 2018-05-15:
There's possibly something else afoot then. Might need debug logs.

stevendt99, 2018-05-15:
OK, just let me know what I need to do and I will collect the data

keirf, 2018-05-16:
You need to wire up the serial usb adapter to the Gotek while installed in your Memotech, and run a serial terminal on the USB virtual COM port on a PC. The default baud rate is 3Mbaud which I achieve with fake Prolific PL303 no problem, using miniterm.py on Linux. But I can drop the baud rate if there are problems (that itself can cause probs though).

You can power the Gotek from the USB serial adapter, or from the Memotech, but not both.

If that sounds plausible I can send you a debug firmware to try.

The other useful thing is a Saleae clone 8-channel logic analyser.... Don't suppose you have one of those? ;)

stevendt99, 2018-05-16:
I do have a USB-TTL serial adapter, although I have not used it yet, I should be able to collect the debug data through it.
I don't have one of those USB logic analysers, but I do have one of these . . . :-)
http://primrosebank.net/computers/maint/tools/tla714.htm
(I'd need to get it out of the loft though)

keirf, 2018-05-16:
Here's a debug firmware: 
[ff_111_1_3mbaud.zip](https://github.com/keirf/FlashFloppy/files/2008546/ff_111_1_3mbaud.zip)

Might also be worth you zipping and attaching your FF.CFG here to have a quick look at.

stevendt99, 2018-05-16:
Thanks, I will give it a try
FF Config File attached
[FF.zip](https://github.com/keirf/FlashFloppy/files/2008718/FF.zip)

stevendt99, 2018-05-16:
I'm struggling to find a terminal program that can run 3M, it's not the speed of the device (device manager lets me select 3000000), but the programs that I have tried so far seem to be limited to multiples of the normal serial port baud rates. The closest that I get is 230400.
I am running Windows, I can't find a Windows build of miniterm, but still looking . . .

stevendt99, 2018-05-16:
Sorry, I've not been able to get a 3M serial program working. Could you do one with a baud rate of 230400 please - if that will be adequate for debugging?

keirf, 2018-05-16:
Here you go:
[ff_111_1_230400.zip](https://github.com/keirf/FlashFloppy/files/2009435/ff_111_1_230400.zip)

stevendt99, 2018-05-16:
Thanks for that. Lots of problems along the way, including failed laptop battery <sigh>, but we're getting there. 
Have flashed the drive. 
Had problems getting the serial port to open, it has now, though no output yet.
Are the Tx and Rx terminals marked on the wiki the Gotek functions, or where the serial adapter pins connect. That is, do I connected Tx on the adapter to Rx on the Gotek or Tx to Tx?
Should the serial messages start by default?

keirf, 2018-05-16:
Tx to Rx, Rx to Tx, the serial messages should start at power on.

stevendt99, 2018-05-16:
OK, There's a problem with my setup then, connected that way, the PC can't talk to the port. 
I will try again in the morning, but I'm not confident.

stevendt99, 2018-05-17:
Not having any success so far, tried with Windows 10/64 and XP, both running Coolterm.
The serial adapter appears to work OK with Tx & Rx looped back together, but can't talk to the Gotek
Have tried powering the Gotek both from the system and the serial adapter (not at the same time), no difference

stevendt99, 2018-05-17:
Still nothing.
Can you confirm that I should not expect to see any indication in the FF version info that I am running a debug firmware?
Can you confirm that the comms settings should be 8-N-1?
If the serial line was working, what would FF be sending on power on?

keirf, 2018-05-18:
Apologies, somehow I built a non-debug firmware (not the first time either) :( Please try:
[ff_111_2_230400.zip](https://github.com/keirf/FlashFloppy/files/2015743/ff_111_2_230400.zip)

If you run dump the dfu or upd file in this zip you will see debug messages within, such as "Attached device %02x:%02x is not supported". If those are missing, you know I messed up ;)

With this firmware you should see banner text immediately on power up, at 230400,8n1.

stevendt99, 2018-05-18:
Ah, now that makes a difference :-)
As you can see, I started a format and it failed
System generated the format failure message at this point
FF Stopped at T23.0

[CoolTerm Capture 2018-05-18 11-54-09.zip](https://github.com/keirf/FlashFloppy/files/2016588/CoolTerm.Capture.2018-05-18.11-54-09.zip)

stevendt99, 2018-05-18:
Yes, I can see those occasional >300us writes.
I will try some other sticks, that one is a cheap (unbranded British Museum souvenir) and upload the results shortly.

stevendt99, 2018-05-18:
Here is a selection of Flash Drives . . .
A Sandisk 64GB (probably USB2) FAT32
A Sandisk 65GB USB 3 (not the one that I used originally) FAT32
A Cheap MicroSD 128MB in an adapter FAT32
A HP 4GB FAT32
These two 64GB drives formatted successfully 5 times in a row
The 128MB Drive failed on the 3rd Format
The HP Drive failed to format twice in a row
[SanDisk64GB.zip](https://github.com/keirf/FlashFloppy/files/2016988/SanDisk64GB.zip)
[SanDisk64GBUSB3.zip](https://github.com/keirf/FlashFloppy/files/2016995/SanDisk64GBUSB3.zip)
[MicroUSB128MB.zip](https://github.com/keirf/FlashFloppy/files/2016997/MicroUSB128MB.zip)
[HP4GB.zip](https://github.com/keirf/FlashFloppy/files/2017002/HP4GB.zip)

stevendt99, 2018-05-18:
For completeness, here's the SanDisk64GB that I was using originally (it was pretty much brand new, as are the other 64gb drives). As you can see, this one failed to format 3/3 times
[SanDisk64GBUSB3_First.zip](https://github.com/keirf/FlashFloppy/files/2017128/SanDisk64GBUSB3_First.zip)

keirf, 2018-05-20:
I have to conclude that the Memotech is very sensitive to write latencies. With `index-suppression = no` FlashFloppy inserts extra index pulses in safe places to minimise time from end of write to next index pulse. 

If you search for six-digit latencies in your logs you will see that all your failing USB sticks have worst-case latencies in the 100s milliseconds. Possibly because they do not like 256-byte writes, as much as anything. Unfortunately batching is non-trivial because the physical layout is interleaved and so sectors don't get written back in the order they are written by the Memotech host. I may improve this in future in other ways (eg. implement a buffer cache).

The MicroSD is interesting, I only see worst case latencies of 70ms or so there, yet that also seems bad enough to occasionally fail a format. That's surprising. I believe the disk-ready circuit in the FDX uses LS123 chips with RC delay circuits -- the capacitors used will be electrolytics and given their age they could be dried out, out of spec, and hence the delay is shorter than it should be.

Generally SD cards have been more predictable in write performance than USB drives. This could explain greater reliability on HxC SD.

keirf, 2018-05-20:
Updated the Memotech section of the Wiki.

closed 2018-05-20 by keirf