keirf/flashfloppy #113: Error E34 after switch back to 7-segment display

BinaryLighthouse, 2018-05-17:
Hi,

Using a Gotek with flashfloppy_v0.9.19a and an OLED display.  All was working fine with images in different folders.  That is until this evening when my OLED died.  As a temporary measure I switched back to the original 7-segment display, only to encounter error E34.  It started working again once I'd placed a few images back in the root folder.

Is this a bug or by design?

tenuke, 2018-05-18:
yea well .. :)
https://github.com/keirf/FlashFloppy/wiki/Error-Messages

E34 = No images found on usb-stick

keirf, 2018-05-18:
Subfolders are only accessible using OLED/LCD displays. Updated Error Messages wiki page to make this clearer for E34.

closed 2018-05-18 by keirf

BinaryLighthouse, 2018-05-18:
OK, that's what I suspected, just looking for confirmation.

> Subfolders are only accessible using OLED/LCD displays. Updated Error Messages wiki page to make this clearer for E34.

BinaryLighthouse, 2018-05-18:
@tenuke 
I already knew that, but prior to today, the wiki didn't state that images in folders are only accessible when using OLED/LCD displays.

> yea well .. :) https://github.com/keirf/FlashFloppy/wiki/Error-Messages E34 = No images found on usb-stick