keirf/flashfloppy #114: .ssd Support extending disk image.

prime6809, 2018-05-18:
A lot of the Acorn .SSD images that are floating about are truncated images where the file is only the bare minimum for the sectors with data on, indeed this is pretty much the default for BBC emulators which create them this way. This is a problem when switching from an emulator to the real hardware and then trying to write on the real hardware, as the data that would be beyond the end of the file is thrown away. 

So it would be good if the file could be extended in these cases so that valid data can be written. This could be done on opening the disk image as I believe the first two sectors of a DFS disk have a couple of bytes detailing the sector count. So should be possible to check this and if the filelength is less than sector count * 256, then extend it to that.

Cheers.

Phill.

keirf, 2018-06-05:
Currently SSD is always opened as C-H-S = 80-1-10, DSD opened as 80-2-10. I can auto-expand to that size, or I can do better if someone tells me what bytes to check (also whether there is a signature or something to test in those first two sectors to see if the volume header can be trusted).

keirf, 2018-06-06:
Here is a firmware for you to try. The image files must be writable (make sure FAT read-only attribute is not set). Auto-extend is configurable in FF.CFG but defaults to enabled.

[ff_114_1.zip](https://github.com/keirf/FlashFloppy/files/2075919/ff_114_1.zip)

keirf, 2018-06-06:
Oops, needed more work. Try this one instead:

[ff_114_2.zip](https://github.com/keirf/FlashFloppy/files/2076035/ff_114_2.zip)

commit 806d4b2e (2018-06-06): image: Auto-extend truncated images (types SSD, DSD, TRD) at mount time.

Configurable (extend-image = yes | no)

Fixes #114
closed 2018-06-06 by keirf