keirf/flashfloppy #115: Support for NEC PC98 1.2MB format

tdaede, 2018-05-19:
This is a 1024 byte sector, 8 sector, 360 rpm format. I think adding support is as simple as adding the following line:

{ 8, 2, 84, 1, 3, 1, 0, _C(80) }

to img.c, but I haven't tried yet and maybe someone else could point me to where I can derive the other values (like gap3), and whether FlashFloppy supports 1024 byte sector formats at all. I'm also a bit concerned it could easily throw off autodetection of other formats.

There are some PC98-specific image formats I can add too, where I don't need to rely on size autodetection.

tdaede, 2018-05-22:
I have a non-working WIP branch here: https://github.com/tdaede/FlashFloppy/tree/pc98
(I got a directory listing once, so I think it's close... will pull out the scope tomorrow)

keirf, 2018-05-22:
You need to adjust stk_per_rev too. Else the bitcell size is wrong (ticks_per_cell).

keirf, 2018-05-22:
Please try the branch here: https://github.com/keirf/FlashFloppy/tree/pc98

tdaede, 2018-05-22:
I updated your branch to complete the host=pc98 support, though the behavior ends up being the same (can do directory listing but not read files, trying to boot gets me an I/O error... but I think that means it read the boot sector). I had to rewire the drive to mate with the laptop-style 26 pin connector on my FD1138C, I'll double check that.

Yeah, there are two other common floppy sizes for pc98 (DD and 300rpm HD IBM format), but I have yet to test them. I'll want to switch it to reading the fdi header, but it's rather... underdocumented (note it's also unrelated to the more common Amiga fdi format)

tdaede, 2018-05-23:
OK, got it fully working, turns out that it needs to be set to Shugart interface (at least the way I wired it). I'll try to do some cleanups and then open a PR.

keirf, 2018-05-23:
Please then also suggest some text for the Host Platforms wiki page.

keirf, 2018-05-25:
I pulled in your merge request but I'm not sure other than 1.2MB format can be correct. The 2DD formats do not fit in a DD track at 360rpm, 250kbit/s. Could it be that the data rate for 2DD is 300kbit/s? And are you sure the correct 1.2MB data rate is 250kbit/s (could well be, it does fit in the HD track no problem)?

Do you know the exact raw img file sizes to be supported? As I know some are 77-cyl right? It may be that using the img_type struct is not a good fit, if we need to adjust rpm and datarate depending on IMG file size. So instead we might need a pc98img_open() which does a switch(filesize) and sets certain params directly, like we do in opd_open() for example.

tdaede, 2018-05-25:
I only tested the 1.2MB format. I had thought the DD formats are 512 byte sectors at half the rate, but I have yet to find any DD images in the wild to test. I can try constructing one to see if I can format it... if I can find some real DD floppies I can also measure the data rate on them. 1.2MB definitely works, so the data rate is at least close enough (I can measure it too, to be sure).

All images I've seen have been 77-cylinder, in FDI or D88 format. I added support for raw images for convenience when imaging floppies with dd and for using tools designed for IBM format floppies.

I considered adding rpm into img_type, but it seemed not worth it for just one host. The current logic has only 360 rpm formats in pc98_type. When img_open falls through those it switches back to the generic list and 300 rpm (hopefully, my particular machine's BIOS doesn't support 300rpm so that's also untested, unfortunately)

keirf, 2018-05-29:
Let's leave it here for now then. If a wider variety of image geometries come to light we can fix any issues that arise as and when.

closed 2018-05-29 by keirf