keirf/flashfloppy #117: FDD/FDD3000 disk format

chapas, 2018-05-23:
Hello Keir,
i was wondering if it would be easy to add support for the Timex FDD/FDD3000 disk format:

140K for SS DD 40 tr
640K for DS DD 80 tr

quoting: "https://8bit.yarek.pl/computer/timex.fdd3000/fdd3000d.html"

What's you opinion?

Regards,

keirf, 2018-05-23:
Just need to know the disk geometry for these formats (primarily: sectors per track, sector size, sector numbering)

chapas, 2018-05-26:
I guess this has the answer you are looking for:

[](http://timex.comboios.info/tos.html)

Can you check the above link?

chapas, 2018-05-26:
( my mistake, sorry ) 

Here is the link: 
http://timex.comboios.info/tos.html

keirf, 2018-06-05:
I believe the FDD3000 track format is 16 * 256-byte sectors per track. This is somewhat like Acorn ADFS. Could you try renaming an image file to have suffix .ADL and see if that then works for you? If not then the sector numbering may be wrong and I can provide a patched version of the firmware for you to test.

keirf, 2018-06-13:
Any update @chapas?

chapas, 2018-06-13:
Hi,
no update, i think you can close this issue for now.
I have an FDD3000 unit, and i have some 3'' disks for it; But the problem is that i can't get the floppy drives to work, i really don't know where the problem lies (hardware maybe). For this i don't know what to do to give you the needed information.

keirf, 2018-06-13:
As I said, I think I know the track layout, so can you try with Gotek and see if renaming an image to have .ADL suffix works?

keirf, 2018-06-20:
I'll close for now. I'm sure this can be sorted in due course.

chapas, 2018-06-20:
Ok, i'm unable to test the firmware. I'm not sure if my FDD3000 is OK, since i can't read any disk on it.
Maybe on a near future, thank you anyway.

chapas, 2018-06-22:
Do you think that a disk image ( DSK Extended ) with 171K created with CPCDiskXP tool, might work with the firmware that you where about to provide?

Encarnado, 2018-06-24:
I have a HFE disk image of 140K and 640K TOS disk.
download it here: http://www.encarnado.com/ssc/TOSdiskHFE.rar
I think I can try later with a CP/M disk. I think I've never FORMAT a 3.5" disk in CP/M but it works the same way as TOS since TOS can list CP/M files on a disk.

keirf, 2018-06-25:
@Encarnado thanks, please can you check if these 160k (not 140k, right??) and 640k images work if you rename their suffix to *.ADL
This will make them be detected as Acorn ADFS format, which happens to be very similar or even the same (16 sectors per track, 256 bytes per sector). The only weirdness I see in your 640k HFE image is that all sectors are reported as side 0, even on side 1! Never seen that before, and I don't know if it is necessary, or what.
If 160k works for you and 640k does not, we can always try a firmware patch to get the side numbering to match your HFE...

Encarnado, 2018-06-25:
Yes, it is 160K disk but in "TOS world" those are refered as 140K because that is the available space when empty. If you have read the page about TOS on my website, the first 16K are TOS (tracks 0 to 2), track 3 is the map (FAT), so 20K are not usable.
The 640K disk is basically the same but logically side 0 have tracks 0-79 and side 1 have tracks 80-159 so yes, it's possible that sectors are reported as side 0.
I think Timex didn't had time to complete support for double sided disks since those were very expensive so TOS A.2 formatting them is through an undocumented option. 
I can't test the .ADL extension since I don't have HxC hardware.

keirf, 2018-06-25:
You don't need HxC hardware/firmware to test the ADL extension -- FlashFloppy on Gotek hardware is what's needed.

Encarnado, 2018-06-25:
I don't own any floppy emulator/replacement. I only have real floppy drives.
I only supplied the disk images because Chapas asked my help and since that, why not help implement the 640K disk format? 
It would be amazing to access 2.4MB of data on a 48K machine 30 years ago!

keirf, 2018-06-25:
Ok well I'll leave this open again for a while. It still just needs someone with a working FDD3000 and a Gotek.

chapas, 2018-06-25:
Thanks keirf and encarnado, I will try to provide the Gotek for the tests flashed with the latest available version of flashfloppy, in case we can find anyone willing to test it.

chapas, 2018-07-02:
Hi,
I bring good news, with the help of a friend that had a FDD3000 unit (didn't know he had one) we've successfully replaced drive B with a Gotek with FlashFloppy (jumpers on JC and S1), we've manage to format (command from the CPM disk) both images provided by encarnado, but while trying to copy (command from CPM disk) it yield failed with "hardware error", yet we were still able to list the contents on drive B, so something got from drive A to B.

This also brings another question, on the FDD3000 the original drive A has a terminator IC but drive B doesn't. Is this something that we need to look in the future if we are add a TOS and CPM image on the gotek and used it on drive A, would it work? Is there some way to emulate a terminated drive or unterminated drive?

Regards

Encarnado, 2018-07-03:
All TOS A.2 discs have TOS and are bootable (as long as the first 4 tracks are not defective). You don't need CP/M for them.
I need to read about the green IC on drive A again.
Have you tried to load something from drive B?

keirf, 2018-07-04:
@chapas Terminators shouldn't matter much, the Gotek has fixed 1k terminators, it's not a great load on the host.

Did you do your tests successfully with IMG (or ADL) images, or using HFE images?

chapas, 2018-07-04:
@keirf All tests were performed with the disks/files provided by Encarnado, meaning HFE's.
I found some more information on google groups about the disk geometry:
https://groups.google.com/forum/#!topic/comp.os.cpm/ybqLHgVdpo0

@Encarnado Yes i did, some of the files (commands) were copied completely and we were able to executed them.

This is a picture of the terminator chip on drive A:

[image]

keirf, 2018-07-04:
Attach HFE images here and I can work out the exact geometry and get IMG support for you. This will likely work more reliably.

chapas, 2018-07-04:
The used disks are the ones created by Encarnado:
http://www.encarnado.com/ssc/TOSdiskHFE.rar

chapas, 2018-07-05:
I think this is it: http://img.chipfind.ru/pdf/cts_corporation/760_761_series.pdf

Encarnado, 2018-07-05:
There is something you should know, as you might realise, Polish people developed a lot of software for the Timex Computer/FDD/FDD3000 because it was the computer they only had for a long time. They used 360K 5.25" drives and since those disks cannot be flipped as the 3" ones, they added a switch to side select signal so they can use both sides. Switch is off, it reads side 0, switch is on, it reads side 1. This poses a lil problem because the disk rotation becomes "backwards" if you "flip it".  Let's say you have a "polish 360K DSDD" disk image and want to read side 1 of the disk to be saved on side 0 of a real disk. TOS cannot read it. That is why there is a "TMX2  ZX Spectrum - TIMEX FDD (side 2)" disk definition  for 22DISK program.

keirf, 2018-07-05:
Okay, please try the attached images as direct replacements for the HFE images you previously tested. The odd suffix is because the physical layout happens to be the same as Acorn ADFS disks. Interested if these read/write more reliably than HFE.

[FDD.zip](https://github.com/keirf/FlashFloppy/files/2166746/FDD.zip)

chapas, 2018-07-12:
Hi,
i have used your ADL files (TOSdisk140k.adl and TOSdisk650k.adl), formatted them both with success on the FDD3000.
I have then made 4 disks copies to 4 (previously formatted) TOSdisk140k.adl  with success, try them out and worked as expected.

I was then trying to use these ADL disk images on FDD3000 Emulator (build upon FUSE) on Windows but without any success, which is by the was expecting a DSK file.

Using a util named TAP2TOS which i got from [TAP2TOS](https://sourceforge.net/projects/fdd3000e/files/tap2tos/) , i was able to generate a DSK file (that I'm attaching) with just TOS. Using this file on FUSE, it worked just fine but it doesn't do much.
I still haven't got the time to try this disk on the FDD3000 but i will ASAP.

Do you know a tool that allows ADL2DSK file conversion?
I guess the DSK might be a more suitable format to hold FDD disks than ADL. Can you check the geometry of the attached DSK disk image? Do you think it might work with the flashfloppy firmware?

[FDD_2.zip](https://github.com/keirf/FlashFloppy/files/2188499/FDD_2.zip)

keirf, 2018-07-12:
@chapas Is DSK the standard image type used for FDD3000? If so then there is no work for me to do -- you should find DSK images work just fine. And if so, we can close this ticket, right?

chapas, 2018-07-12:
OK,
i will test it ASAP, you can close this issue if you wish.
One more thing can you create or teach me how to create a 640K DSK floppy disk image with similar geometry found on 140K floppy image DSK one?

keirf, 2018-07-12:
I think it is possible with the hxc software but one way or another I will make you a 640k dsk image tomorrow.

keirf, 2018-07-14:
This was possible with "HxCFloppyEmulator software". Instructions:
1. Select "Load Raw Image". Settings: IBM MFM, Bitrate 250000, RPM 300, 80 tracks, 2 sides, 16 sectors per track, 256 bytes per sector, Sector ID Start 0. Everything else default.
2. Select "Create Empty Floppy"
3. Select "Export". Choose CPC DSK as export type.

keirf, 2018-07-14:
Here is a 640k DSK image for you to try formatting:

[640k.zip](https://github.com/keirf/FlashFloppy/files/2194663/640k.zip)

keirf, 2018-07-15:
Anyway, since DSK format seems to be used and since that is already well supported I will close this ticket. Any new issues arise, reopen this ticket or raise a new one, please.

closed 2018-07-15 by keirf