keirf/flashfloppy #120: Sega System 24 support

TheRealApocalypse, 2018-05-29:
Hi,
I found flash floppy working fine for most Sega System 24 games (arcade).
Working games share the following format:
	// Sector  Size
	// 1       2048
	// 2       2048
	// 3       2048
	// 4       2048
	// 5       2048
	// 6       1024
	// 7        256
	track_size = 0x2d00

Non working games have this format:
	// Sector  Size
	// 1       8192
	// 2       1024
	// 3       1024
	// 4       1024
	// 5        512
	// 6        256
	track_size = 0x2f00

In any case the .img (sometimes named .s24) have to be converted to .hfe format first.
Please find attached a working hfe file (good) and a non working one (bad).

[S24_HFE.zip](https://github.com/keirf/FlashFloppy/files/2046615/S24_HFE.zip)

Regards,.

keirf, 2018-05-29:
It might be read throughput limitation biting again, since FlashFloppy has no buffer cache it may fail to keep up with the required data rate for reading a high-density HFE image. I do need to add a caching layer at least for metadata, but equally I want to get to v1.0 first before making such major changes.

The easiest solution now is perhaps to support .s24 directly. Is .s24 the accepted suffix or should I support .img too? (Will need to add a `host = system-24` config option to detect .img).

TheRealApocalypse, 2018-05-29:
Hi,
Floppy images, used by emulators for instance, have the .img extension.
I think it was simply renamed .s24 by Jeff from HxC to avoid confusion with IBM image files.
So I'd say both extensions are likely to be used but just go with what's the easiest to implement for you.

Regards.

TheRealApocalypse, 2018-05-30:
Hi Keir,
Hold everything ;)
I've discovered the HxC conversion tool doesn't create valid files for the problematic games. I've manually fixed the hfe file and now it works flawlessly with flash floppy.
I will try other problematic games and let you know the outcome.

Regards.

keirf, 2018-05-30:
Aha :) They looked superficially valid, correct sector sizes at least. The tracks were rather long, so time between index pulses would be extended, but most systems don't care too much about that.

keirf, 2018-05-30:
Which header value(s) did you have to fix up?

TheRealApocalypse, 2018-06-02:
Yes sorry, Jeff from HxC has patched his software but I wasn't using the latest version and now all games work perfectly.

keirf, 2018-06-02:
Excellent, no work to do here then :)

closed 2018-06-02 by keirf