keirf/flashfloppy #121: Self describing MFM/FM RAW image formats

staticmem, 2018-06-01:
As a feature request I would like to see RAW disk images using a self describing disk image file naming format added.  By RAW I mean headless disk images that consists of only sector data and no other embedded information about the disk layout.  RAW images are typically arranged in an alternating format so that double sided disks have a file structure of T0H0, T0H1, T1H0, T1H1, etc. for double sided disk.

Basically it would be mainly intended for normal non copy protected disk formats.  Any disk formats that are more involved such as physically skewed sectors on the tracks themselves might need more options to describe.

The naming structure would consist of something like "diskname.C-H-S.raw".  Using this format would assume a default format of 512 bytes per sector.  A further option could be specified if the sector size is not 512 so a 256 byte sector disk could use "-z256" or to keep the name short use a physical shift value instead such as "-p0, -p1, -p2, -p3" (128, 256, 512, 1024).  The default format would assume MFM but could add "-f" for FM and "-m" for MFM.
If the data rate needs to be specified too then something like "-rnnn" could be added otherwise use a default value of 250 kHz.

Any GAP values required can be calculated from the values shown above but could set it with another parameter of say "-gnnn".

For the first sector number if 1 is not used could use "-snnn".

For my use I'd say over 98% of my disks would use the default C-H-S format.  I guess the FF config file could also have default settings for the other values.

keirf, 2018-06-05:
Yes this is a good idea. I will implement it.

staticmem, 2018-06-05:
It could also save a lot of work for you keirf if a simple optional macro file support was added.  For example with a Microbee raw format disk that has a filename of say "my_disk.ss80" I know from the "ss80" part that the format is 80 tracks, 1 head and 10 s/t.  A user defined Macro file containing some entries like the following would be great:

.ss80     .80-1-10.raw
.ds80     .80-2-10.raw
.ds40     .40-2-10.raw

In the case of the example given it would be processed to be "my_disk.80-1-10.raw"

keirf, 2018-06-05:
Also a good idea ;)

keirf, 2019-03-06:
See also discussion in #207

keirf, 2019-03-12:
Okay, here is a firmware to test. Please note the documentation in examples/IMG.CFG. You need to define your own suitable IMG.CFG and tag your IMG/IMA files with the tag(s) you define.

[ff_121_1.zip](https://github.com/keirf/FlashFloppy/files/2957334/ff_121_1.zip)

alank2, 2019-03-12:
That looks pretty sweet Keirf!  Will check it out.

staticmem, 2019-03-13:
Thanks keirf.  I will try it out and post back.

staticmem, 2019-03-13:
So far I've just looked at the IMG.CFG file.  Would it be possible to have another option to specify a sector offset value to be used from a specific cylinder number and up.  i.e: A disk that uses physical sectors 1-10 on the first 2 cylinders (both sides) then afterwards they have an offset of 20 so are numbered 21-30 for cylinders 2-79 for an 80 cylinder double sided disk.

keirf, 2019-03-13:
Yes I'm not sure how super configurable to make this at the moment. But some different parameters at least for cyl 0 may make sense. I shall add that. I don't promise to maintain IMG.CFG compatibility for these 'extended geometry features' -- depending how much flexibility is needed I might come up with a completely different more flexible way to define extended geometries in future.

keirf, 2019-03-13:
Okay a new firmware with new features:
1. You can specify a catch-all 'default' tag which will catch even untagged IMG/IMA files
2. You can specify sector IDs for cylinder 0 separate from cylinders 1+

[ff_121_2.zip](https://github.com/keirf/FlashFloppy/files/2960596/ff_121_2.zip)

alank2, 2019-03-13:
Fantastic!  The #1 option means setting it once and forgetting it.

Simone2018, 2019-03-13:
Hi, keirf, thank for this option:
I made a DD floppy with 254 tracks... with :
cyls  = 254
heads = 2
secs  = 9
bps   = 512
id    = 0
but is unreadable (but recognized by FF: show T00.0)
my floppy is a 720kb (so 250kb/s),
Can you add an option for 250/500 kb/s (DD/HD) speed?
thank you

keirf, 2019-03-13:
@Simone2018 the parameters you used should have worked (although shouldn't id be 1 for PC format?).

I can test this tomorrow, I've not actually heard anyone have success with this new feature yet....

SilverGreen93, 2019-03-13:
Wow, thanks a lot @keirf for this update! I tested firmware ff_121_2 and it works great for supporing HC2000 disks. I just defined a custom format and works a treat!

dandoore, 2019-03-13:
@keirf I'm having no joy with the Osborne disks at all, I have uploaded the IMG's HFE's and CFG in use here if that helps:  [Download](http://www.dandoore.com/flashfloppy/IMG-test-Osborne.7z)

keirf, 2019-03-13:
Here is an updated firmware which
1. Raises an Error (31 = Bad Image) if a bad tag is defined
2. Makes the 'id' parameter optional (default value 1).

[ff_121_3.zip](https://github.com/keirf/FlashFloppy/files/2963049/ff_121_3.zip)

keirf, 2019-03-13:
> @keirf I'm having no joy with the Osborne disks at all, I have uploaded the IMG's HFE's and CFG in use here if that helps: [Download](http://www.dandoore.com/flashfloppy/IMG-test-Osborne.7z)

@dandoore your download link doesn't work :(

dandoore, 2019-03-13:
> @dandoore your download link doesn't work :(

Bollocks - try this https://www.dropbox.com/s/8ytnb8qork2457s/IMG-test-Osborne.7z?dl=0

keirf, 2019-03-13:
> try this https://www.dropbox.com/s/8ytnb8qork2457s/IMG-test-Osborne.7z?dl=0

That worked. I'll have a look.

dandoore, 2019-03-13:
@keirf  Ah - with the ID not specified that's a lot better - let me test some more!

keirf, 2019-03-13:
Yes, the IDs on the occ1 and occ4 formats start at 1. So the new default 'id' value is correct. Those disk images should now work fine.

osb1sssd won't work because I don't yet support FM mode with Index-Address-Mark, so those images will fail to load (not that you supplied any). I will fix that tomorrow. But for now if you have such images, just try setting iam=no for that tag.

Finally, the supplied OSE-CPMA.img has no tag, and will match the default tag, which also presumably has the wrong 'id' field. Again you should be able to remove explicit setting of 'id' and it should work.

Thanks!

dandoore, 2019-03-13:
> osb1sssd won't work because I don't yet support FM mode with Index-Address-Mark, so those images will fail to load (not that you supplied any). I will fix that tomorrow. But for now if you have such images, just try setting iam=no for that tag.

Just tested some SSSD FM IMG for the Osborne1 with that and that works with iam=no. \o/

> Finally, the supplied OSE-CPMA.img has no tag, and will match the default tag, which also presumably has the wrong 'id' field. Again you should be able to remove explicit setting of 'id' and it should work.

Yup, that was the test for the [default] - I've fixed that up to ignore ID and all working.

Cheers for all your help :-)

keirf, 2019-03-13:
Further update which supports mode=fm and iam=yes (ie. FM mode with Index-Address-Mark):

[ff_121_4.zip](https://github.com/keirf/FlashFloppy/files/2963506/ff_121_4.zip)

witchy69, 2019-03-13:
I've just tried a 40T DSDD image with a 2:1 skew made from a Newbrain CP/M 2.2 disk and I'm getting 'no response from controller'. I know the FF.CFG is OK because I can select the same image but in HFE format and it boots. 40T, 2 heads, spt 10, bps 512. That's with the latest firmware you've just posted :) 

[nb-cpm.dsdd40.img.gz](https://github.com/keirf/FlashFloppy/files/2963533/nb-cpm.dsdd40.img.gz)

keirf, 2019-03-13:
@witchy69 you need a smaller GAP3 for 10 * 512b sectors. Try `gap3=30`.

witchy69, 2019-03-13:
Sweet! That works not only with the image I've just created but also the 80T image I was working with that I mentioned on facebook yesterday (Adrian here). 

Nice one. I've never found a decent explanation of the GAP3 though I must admit it took me a while to get my head around skew and interleave. Thumbs up!

keirf, 2019-03-13:
Gap3 is the gap after the sector data. It is needed on a real floppy disk because sectors get overwritten one at a time, and drives rotate at different speeds. Without a post-data gap, if the track was formatted on a slow-spinning drive (so the sectors are shorter on disk) and then a sector overwritten by a faster-spinning drive (so the new sector is longer on disk) then the sector would overwrite the start of the following sector! So standard Gap3 values are designed with the extremes of motor variations in mind.

Reduced gap3 values allow more data to be squeezed into a track, and in extreme cases (if a disk is known to be read-only) can be reduced to zero.

keirf, 2019-03-13:
A further update which supports rate=xxx to specify a data rate in kHz.

[ff_121_5.zip](https://github.com/keirf/FlashFloppy/files/2963680/ff_121_5.zip)

witchy69, 2019-03-13:
Ah right. This goes along with something else I learned last  year which is floppies written on an 80T drive double stepping might not read on a 40T drive because of the thickness of the heads. I spent the 80s thinking floppies were really easy things!

Image update, I've added the gap3 value to dsdd80 and renamed all my other 80T images - they're working too.

staticmem, 2019-03-13:
@keirf: My "uBeeDisk" program does auto GAP calculation based on WD279x format specifications for FM and MFM.  It's GPL License code if you to make use of some of it.

The release topic which includes a download link for sources and binaries here:

https://www.microbee-mspp.org.au/forum/viewtopic.php?f=22&t=2033#p11065

keirf, 2019-03-13:
Thanks, looks sensible to set the GAP to zero, calculate track size, and then divide the remaining available track space by number of sectors, perhaps up to some max GAP value.

I may implement that.

Simone2018, 2019-03-14:
Great Job Keirf! 😊
Now with default ID and data rate of 250kHz, work flawless! 
I finally have DD dos disk of 4.5 Mb.... I'll continue to test more! (Before I had to convert to HFE and it took a lot more space)

keirf, 2019-03-14:
This is now released (v2.0a). Closing the ticket. Please open new tickets for bug reports or further feature requests.

closed 2019-03-14 by keirf