keirf/flashfloppy #122: Support for all IBM PC floppy formats

CarlosTex, 2018-06-01:
Hi,

i'm an IBM PC retrogamer and aficionado. I currently have several DOS machines, each with capability to use a Gotek floppy emulator. Currently i have been using the default firmware with all its limitations but currently i have dscovered flashfloppy. So i programmed flashfloppy, and realized that not all formats are supported. It seems that 1.44MB and 720k are supported just fine, but i would like to see support for all formats. This is important in the IBM PC world as some bootable games used 160K, 180K, 320K, 360k, 400k (Electronic Arts titles). I noticed i could run 1.44MB IMG files just files, but none of the booter games with the older formats work. Instead i get the ERR*31 on the OLED.

I hope you consider this, otherwise this excellent firmware won't be of much use to IBM PC retro enthusiasts.

keirf, 2018-06-01:
Do you have any info on geometry for these formats (sides, tracks, sectors per track, sector size) or alternatively attach some HFE images to this ticket if you can. You can convert IMG to HFE using HxC software and they would then work with FlashFloppy. But I can also add direct IMG support.

CarlosTex, 2018-06-01:
OK i can do that:

FORMAT    Sectors per track     Tracks per side    Sides      RPM
160k                     8                           40               1          300  
180k                     9                           40               1          300
320k                     8                           40               2          300
360k                     9                           40               2          300
400k                    10                          40               2          300        (Electronic Arts Booters)
1.2MB                  15                          80               2          360
All of the above are 512 bytes per sector.

I assume you have support for the 1.2MB format but i included data for it as well.

I don't know much about .HFE, i only started adventuring through these stuff today. I'm know a whole lot about IBM PC, but i know little about other machines and HxC stuff and the HFE files.

If you can add .IMG direct support that will be fantastic, because tons of archived images will work immediately. 

There were a couple of other file types used in the IBM PC world which were .TD0 created by a program called Teledisk and .CP2. CP2 was a file format created by special hardware, an ISA card which connected to floppy drive so you could copy disks with copy protection. This is not something i would worry for now, but maybe this would be something nice to add in the future. I know the a guy who knows the .TD0 file intimately (he worked on it).

For now, if you can add support for the formats above, flashfloppy will be covering 99% of all that is necessary in the IBM PC world. There are other formats, like those extra high density 2.88MB, but those were not really used a lot. 

I'll be waiting for a new firmware update, i'll test it immediately and give some feedback. I'll be anxiously waiting! Thanks for the fast answer BTW!

Regards,

Carlos

CarlosTex, 2018-06-01:
By the way if you want to consult a list with everything the Wikipedia page has a lot of info:

https://en.wikipedia.org/wiki/List_of_floppy_disk_formats

keirf, 2018-06-01:
These will be easy to add so long as they don't collide in file size with any existing 80-track formats.

CarlosTex, 2018-06-01:
Well if you can do an update in a jiffy, i can test them immediately. 

On the IBM PC side there are some low density 1 side formats with 80 tracks which will collide in file size. These are 320k and 360k. But these formats with 80 tracks were really NEVER USED. So there's no doubt that 160k, 180k, 320k, 360k, 400k 40 tracks and 720k, 1.2MB and 1.44MB 80 tracks represent 99.9% of the IBM PC world. If all these are supported by flashfloppy you will have 99.9% compatibility with the IBM PC covered.

CarlosTex, 2018-06-01:
I should add another one for the booter titles. It's the same as for the Electronic Arts titles but one side only:

FORMAT SECTORS TRACKS SIDES RPM
200k 10 40 1 300 (Electronic Arts Booters)

CarlosTex, 2018-06-02:
And suddenly all booter games started to work!! I tested all formats i asked for and all are working. Amazing work! Thank you so much for the fast response and quick fix!

closed 2018-06-05 by keirf

Simone2018, 2018-06-05:
I agree with staticmem, you can find info on boot Sector like this:

>  79 #define BS_JmpBoot 0 /* x86 jump instruction (3-byte) /
>  80 #define BS_OEMName 3 / OEM name (8-byte) /
>  81 #define BPB_BytsPerSec 11 / Sector size [byte] (WORD) /
>  82 #define BPB_SecPerClus 13 / Cluster size [sector] (BYTE) /
>  86 #define BPB_TotSec16 19 / Volume size (16-bit) [sector] (WORD) /
>  87 #define BPB_Media 21 / Media descriptor byte (BYTE) /
>  89 #define BPB_SecPerTrk 24 / Number of sectors per track for int13h [sector] (WORD)/
>  90 #define BPB_NumHeads 26 / Number of heads for int13h (WORD) /
>  100 #define BS_55AA 510 / Signature word (WORD) */ "

staticmem, 2018-06-05:
@CarlosTex:  You mentioned "So i programmed flashfloppy, and realized that not all formats are supported. It seems that 1.44MB and 720k are supported just fine, but i would like to see support for all formats."

There's all sorts of formats that are possible and it really requires the use of the BPB as Simone2018 mentions.  A good description of it can be found here: https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system

It could be forced to look into the BPB by adding a ".dos" to the file image extension perhaps?

keirf, 2018-06-05:
@staticmem this further conversation belongs on #68

commit fd20ebb9 (2018-06-05): img: Support common IBM/PC retro formats out of the box.

Fixes #122