keirf/flashfloppy #125: Non-linear order of image files to slots/slow browsing of slots in high number range

mpc3k, 2018-06-04:
Hi all,

I am using FlashFloppy for two weeks by now, powering a Gotek with OLED display inside of an Akai S950 sampler. The image files I use are sourced on the HxC website (Quick Install Disk Images). On an 8GB USB Stick, that amounts to a total of +2100 images named "DSKAxxxx.HFE". The image files have been renamed subsequently (using an Automator action on OS X).

The only two issues I have is:
#1 The order of appearance in the display. The slot number does not always correspond to the disk image file name. This can be frustrating, as it becomes a hit and miss attempt when you want to select a specific image file.
#2 Browsing image files in the lower number range is significantly quicker, than browsing e.g. the higher end of the number range. The latter becomes painstakingly slow.

ad 1: I have thought about whether this may be related to creating and renaming the image files on OS X. I have yet to try another attempt on a Windows machine. Is this a known issue?
I've looked into the image files with an Hex Editor to check for some kind of index within the files, but it seems there aren't any.

ad 2: Is there a sensible file limit with FlashFloppy? In terms of fast operation?

Cheers, mpc3k

mpc3k, 2018-06-04:
Thanks, Keir. 

Although [FatSort ](http://fatsort.sourceforge.net/)was easy to install via MacPorts, it didn't get proper access to the filesystem, for reasons beyond me. [DriveSort](http://www.anerty.net/software/file/DriveSort/) got the job done quickly. Great! But then ...

A **word of warning** goes to your recommendation of the [Auto-Sort script](https://github.com/discoHR/auto-organize-files-recursively). This script did create a couple of folders for sorting those files, alas, then wiped _all_ image files off the USB stick. :((
Yes, the packaging tells you to _make a backup before using it_. I did not, which I had to regret instantly.
Here l am, recovering the files for hours already, sadly not including those currently used in production, as the first four image files have been erased for good (or bad) by creating a number of empty folders.

keirf, 2018-06-04:
Okay thanks. That is a bummer! I think perhaps I will look into creating my own alternative, in Python.