keirf/flashfloppy #126: Amiga 1200 - not detecting disk changed

ojaksch, 2018-06-04:
I'm using v0.9.19a in an Amiga 1200 w/ Escom board, but removed their mod as suggested in Wiki. Disk operations are fine, but it isn't detecting a disk change.
Any idea?

keirf, 2018-06-04:
I think the retrogamemodz YouTube video mentions a link wire might need to be added, in addition to removing the Escom wire? Apart from that, also make sure there is no jumper at JC at rear of Gotek. Jumper S0 only.

keirf, 2018-06-04:
Also, if you have FF.CFG config file make sure it states `interface = shugart`

ojaksch, 2018-06-04:
Yeah, did this fix today and added this wire (but makes no difference - tested with and without wire). Jumper is at S0 and FF.CFG states `interface = shugart` already.
Your Wiki is fine and leaves no misunderstandings, so that's why I'm asking...

keirf, 2018-06-04:
Do you have a multimeter and could do a continuity test from pin 2 of the Gotek floppy connector to the appropriate CIA pin on the amiga mainboard?

keirf, 2018-06-04:
Problem could even be the floppy cable for example...

keirf, 2018-06-04:
The "appropriate CIA pin" is pin 4 of U7. That is the disk-change signal, which as I said above is output on pin 2 of the Gotek connector.

keirf, 2018-06-05:
Closing. This is a support ticket not a bug ticket.

closed 2018-06-05 by keirf

ojaksch, 2018-06-05:
Sorry for noising around my personal bugless problems with gotek/flashfloppy.
Anyway: **Many thanks** for any tip you gave. It IS the floppy cable. It has twisted wires you couln't see because it's a welded round cable. Exchanged for a flat cable, all problems gone.

keirf, 2018-06-05:
Not a problem, I don't mind the occasional support ticket but I keep the tracker clean after advice is handed out. There is a Facebook griup which is the best place for support but I understand not everyone likes FB.

ojaksch, 2018-06-05:
Full understanding. And yes, no FB here and never will be. This is problematic: Where to ask such questions? Sure, it MAY be a general question to ask anywhere else, but IS related to flashfloppy - or COULD BE. That's why I asked here at first.
However, solved by your friendly advice. Again: Thanks! :-*

keirf, 2018-06-05:
There is a forum http://www.primrosebank.net/flashfloppy/ but it is rarely attended. Facebook has sucked the life out of most old-style forums.

Squirrel61, 2018-06-05:
@Larry: Create a Fakebook account. I did, using fakenamegenerator.com. It
really helps a lot for those pages that actually should be forums instead
of facebook groups. Nothing wrong with the groups, but forums are much
easier to use because they usually have a tree structure, where facebook is
just one huge dedicated timeline.

silelis, 2019-05-23:
@keirf,
I have Escom Amiga 1200 and want to will add Gotek.
I don't want to avoid to make this RetroGameModz and wonder If there is possibile to make fix in FlashFloppy firmware that will make Gotek to Escom compatibility?

keirf, 2019-05-23:
If you don't want to modify your Escom board, place a jumper at position JC at the rear of the Gotek, as well as position S0 (as usual). Note that, as is normal with Escom boards, a lot of track-loaded games and demos will hang at boot. Try out Arte by Sanity (demo) for example. Only way to get it working is to mod the Escom board and remove jumper JC.

silelis, 2019-05-23:
@keirf,

Than You for quick replay. I will consider the "wire mod".

Is this mod also required (on Escom) if I want to use Gotek as DF1?

I want to avoid it and I consider to unplug DF0 if it helps to work gotek as external DF1. But still don't know if there is still track-loaded problem imn this configuration?

keirf, 2019-05-23:
No need for wire mod for gotek to work fine as df1 (after all, escom boards work fine with standard amiga external drives). However most trackloaders will only boot from df0

silelis, 2019-08-22:
I need Your help about Escom 1200 floppy fix.

I have made Escom motherboard rework and floppy rework according to this movie and this tutorials:
https://www.youtube.com/watch?v=G6fYOjTYvXM
http://amigaga.chez-alice.fr/classic/bidouilles/disquette/a1200fix.htm
https://www.tsb.space/knowledge-base/amiga-1200-escom-floppy-fix/

I have removed wire between pins 2 <==> 34 and make shortcut between 1<==>34.
Like bellow:
https://photos.app.goo.gl/3TbhCmcpNENSCSut9

I have also modified PC panasonic floppy:
https://photos.app.goo.gl/hwoSnWdy4CgPfrma6

And as DF0 both modified floppy and Gotek are detected as Amiga devise and works.

The problem is with DF1 detection (both modified floppy and Gotek) are not detected. I checked it connecting devices directly to IDE tape and to DB23 connector.

What is wrong with my Amiga 1200? What should I check additionally?

BTW: In this Amiga I also have 3.1.4 kickstart by Hyperion.

keirf, 2019-08-22:
@silelis You cannot directly attach a real floppy drive as an external drive without some support electronics: this is the small circuit board at the back of all Amiga external drive enclosures. This does the following:
 - Generates the MOTOR signal for the drive
 - Generates the drive-present identification signal for the Amiga

So if you are directly connecting to the internal header or the DB23 with a simple cable (no circuit board) there is no chance the real drive will work.

The Gotek can work because it doesn't care about the MOTOR signal and it will happen to give an appropriate drive identification signal so long as one of the following is true:
 1. There is an image mounted on the Gotek while the Amiga boots
 2. You have `interface = amiga` in your FF.CFG file on the USB stick

silelis, 2019-08-22:
@keirf,

First of all to attache DF1 I have bought this adapter:
https://www.ebay.pl/itm/233144988083?ul_noapp=true

On adapter all "JUMPERS" are on "AM" position and drive "DF1".
The strangest thing is that in case I switch JP1 to "PC" position it boot sequence detects DF1, but Amiga will not read GOTEK.

Can You specify few more things:

1) Generates the MOTOR signal for the drive <-- What do You mead? On Gotek there is jumper "MO" is it connected with this? In my configuration MO is opened.

2) Generates the drive-present identification signal for the Amiga <-- On adapter I have DF1 ad well as on GOtef "S1".

3) There is an image mounted on the Gotek while the Amiga boots <-- adf's are mounted. Gotek works perfectly as DF0.

4) You have interface = amiga in your FF.CFG file on the USB stick <-- I had flashed Gotek with this firmware "v3.2a" ( https://github.com/keirf/FlashFloppy/releases/tag/v3.2a ), also FF.cfg file is from this release.

Any advice?

keirf, 2019-08-22:
Put the jumpers set for AM and DF1 on the adapter, and set the jumper on the Gotek as S0.

Don't worry about settings in FF.CFG, since you have a proper adapter (not that I tested this one myself, but let's assume it works).