keirf/flashfloppy #127: More/alternative button options + in between font

CarlosTex, 2018-06-07:
I would like to propose and ask if it is possible the following options/enhancements:

Additional options for the "twobutton-action" variable:

Button 1 - Previous (-10 slots)
Button 2 - Next       (+10 slots)
Both       -               (Slot 0)

This is great when you have a folder with a couple hundreds of files. Buttons jump back and forth in greater increments and the rotary encoder gives the single digit precision. 

or any other suggestions...

For the "oled-font" i would like to ask if its possible to implement an in between font, like 7x14. This would be a good between Narrow and Wide fonts, ex. Narrow, Normal, Wide

keirf, 2018-06-08:
1. Hold prev or next button and it will accelerate through your slots. This may be good enough?

2. I think two fonts is enough and I never saw a 7x14 font I like. You can suggest one though.

CarlosTex, 2018-06-08:
1. Its not bad but not as precise. If you don't think its worth to implement this its not a huge loss, it would just be a nice addition for people with large number of slots on different folders. Its kind of redundant to have the buttons do the same as the rotary encoder. 

2. I like the following one: https://github.com/chneukirchen/sq

This font is 7x15 though but i like it. I think this would provide a good mid alternative between the Narrow and Wide fonts.

keirf, 2018-06-08:
Congratulations on finding an actually nice 7px font😉

keirf, 2018-06-11:
I experimented with the font a bit and it doesn't really feel wider to me, more that there is more spacing horizontally between characters. However I do like the font it is based on (Codec, by Chris Cannam). And there is a 6x13 version of that, which overall I like better than the standard 6x13. It feels like i uses vertical space better, and is a tiny bit more readable. So I'm switching to that for the 6x13 font.

keirf, 2018-06-11:
The +/- 10 thing is moderately a pain to implement. I could make the accelerated function of the buttons kick in quicker in rotary mode? (eg. give a config option to allow this)?

RetroClinic, 2018-06-11:
How about making the buttons scroll through the 26 letters of the alphabet so the encoder just displayed those files? Wouldn’t be as complicated as an alphabetic sort, but would allow you to search for stuff beginning with a certain letter?

From: Keir Fraser 
Sent: Monday, June 11, 2018 9:30 AM
To: keirf/FlashFloppy 
Cc: Subscribed 
Subject: Re: [keirf/FlashFloppy] More/alternative button options + in between font (#127)

The +/- 10 thing is moderately a pain to implement. I could make the accelerated function of the buttons kick in quicker in rotary mode? (eg. give a config option to allow this)?

keirf, 2018-06-11:
@RetroClinic FlashFloppy will only display things in FAT dir order, it doesn't sort at all. The best way to get what you describe is to pre-process your USB stick: split images into folders by first letter (eg 36 folders 0-9,A-Z), and then run a FAT sorter across all folders. This has the dual benefits of easier navigation, and faster navigation by FlashFloppy (which doesn't like lots and lots of files in a folder).

RetroClinic, 2018-06-11:
No, I understand that, the VNC2 on my DataCentre does the same, what I meant was to do it so that it only shows, say files staring with “E”, if you select “E” from the front buttons. The firmware would keep looking through the current DIR until it saw something starting with “E” then display it, next time the encoder was turned, show the next file that starts with “E” etc.? Doesn’t have to alphabetically sort them, just only show the files that start with the first letter of the letter selected as it comes across them.

From: Keir Fraser 
Sent: Monday, June 11, 2018 10:01 AM
To: keirf/FlashFloppy 
Cc: RetroClinic ; Mention 
Subject: Re: [keirf/FlashFloppy] More/alternative button options + in between font (#127)

@RetroClinic FlashFloppy will only display things in FAT dir order, it doesn't sort at all. The best way to get what you describe is to pre-process your USB stick: split images into folders by first letter (eg 36 folders 0-9,A-Z), and then run a FAT sorter across all folders. This has the dual benefits of easier navigation, and faster navigation by FlashFloppy (which doesn't like lots and lots of files in a folder).

CarlosTex, 2018-06-11:
>@keirf That's nice keir, i'm looking forward to try the new 6x13 font!

As far as accelerating the function of the buttons i think its a good idea (if the +10/-10 is hard to implement, no need for you to lose time on it) and it's better even if you make this user configurable via the FF.cfg file.

@RetroClinic 

The alphabetical order idea is nice, but coding it would be a bit of a pain, as he needs to implements loops to search for the files and selecting them by order. I think this would make FlashFloppy even slower, specially if the USB stick isn't sorted first by the user.

Right now i think the priority is to optimize FlashFloppy going through the files, to make folders with hundreds of files faster to navigate. Like Keir said in another thread, it's doable.

commit ee8beff1 (2018-06-11): FF.CFG: New option "twobutton-action = rotary-fast"
Fixes #127
closed 2018-06-11 by keirf