keirf/flashfloppy #128: df0 - df1 file assignment

kipper2k, 2018-06-08:
Hi,

 I built a new gotek pcb with all the advanced options for second drive select etc. Is there going to be an option that you can "flag" specific images to run in an emulated DF1: for example, assign a game disk 1 to the regular df0: and  assign disk 2 of that game to DF1:

  I guess in order to try to get any benefit of this option then the disk 2 image should be listed in the slot directly after the disk 1 image so that the gotek will automatically look in that slot for the second image, and if it finds the correct image then either assign to df1: if it allows, or replaces disk1 image in df0:

 I know some games are NDOS and would not support DF1: but there are a few out there that would , is this a possibility

Regards

Brian

kipper2k, 2018-06-08:
[image]

kipper2k, 2018-06-08:
oops, closed accidentally

keirf, 2018-06-11:
Are you talking about direct navigation (via OLED display, for example) or image selection via the file selector into pre-assigned 'slots'?

Your board looks very nice, did you connect through motor-on signalling and if so did you do anything about generating DF1's proper motor signal? Not that FlashFloppy uses motor signal at all at the moment, so it's hardly critical...

kipper2k, 2018-06-11:
Hi Keirf,

 I followed your Advanced gotek design changes and added the changes to the PCB for the following, 

- PC12-15 to gnd for board identifier
- PA3 second drive select
- PA15 motor on signal

 What i was thinking for the multi disk games was that if the user assigns them in a sequential order, ie, disk 1 in slot5 and disk 2 in slot6 that if you do enable 2 floppy drive operation in future updates  you can tag disk 2 for df1 in the slot menu so that when loading disk 1 the FW will look to see if the adf in the next sequential slot is tagged as DF1 and if so assign it to DF1 automatically.

 A little confusing trying to explain but basically you would be getting the FW to check if the adf in the next slot has been identified as requesting DF1 and automatically assigning it to DF1 so no user intervention is required. There could be times when this is not needed so an Enable/disable feature in the menu could/would be optional

 If you want me to test some of future updates on the modified board let me know. If you want one of these boards to test with yourself i would be happy to donate one for you so you can play with it. Let me know

Regards

B rian

keirf, 2018-06-11:
If you are talking about the host file selector (AUTOBOOT.HFE) then I think that already understands drive A/B separate slots. That just needs to be configured, possibly at build time.

I think by your talk of slots, that this must be what you mean.

closed 2018-06-13 by keirf