keirf/flashfloppy #130: Rotary encoder - either requires 2 "clicks" or skips by 2 positions

TzOk83, 2018-06-13:
Maybe it is my specific issue, but I've got rotary encoder, which either moves 1 position, every two clicks (normal), or skips by 2 positions (Gray). Actually - when Gray pattern is chosen it moves 1 position every 1/2 click - if I stop in between clicks it moves only 1 position, but when I allow it to click, it moves to the next position.

keirf, 2018-06-13:
Does it have a green base by any chance?

TzOk83, 2018-06-13:
Yest it has... I've just found another closed issue concerning this one.

keirf, 2018-06-13:
Buy another rotary encoder. Maybe I can support this type if I'm sent one. But it's easy enough to find cheap ones that do work.

RetroClinic, 2018-06-13:
I just did a search for "green encoder" and didn't find any results? I've been sent some of these from the same seller that was supplying blue ones, I thought they were the same, but I presume from reading this that they're electrically different somehow?

keirf, 2018-06-13:
The green base isn't a selling point, it's just an observation that these incompatible encoders seem to all have a green base rather than blue. Maybe the green ones can be supported but I'd need one to test with.

RetroClinic, 2018-06-13:
Ok, I'll give one a try on the next set of builds I do, see if I can see a difference in the waveform output. Happy to send you a couple if it helps?

keirf, 2018-06-13:
Yes that would be helpful thanks. I guess confirm yours don't work, first.

raybellis, 2018-06-13:
Even within a single manufacturer's range of encoders you can find models that do one full quadrature cycle per detent and others that do only a quarter cycle.    For example most E-Mu synths use a Bourns ECW0D-C24-BD0009 where the D0009 indicates 36 quarter cycle detents but there's also an E0036 variant which has 36 full cycle detents.  (see https://www.mouser.co.uk/datasheet/2/54/CW1J-777413.pdf)

I'm guessing somewhat, but your encoder sounds like it's a half cycle per detent, whereas FF is expecting a quarter cycle per detent.

keirf, 2018-06-13:
The trace posted in the second half of ticket #50 seems to show a full cycle per detent, but inconsistent phase difference between the encoder pins. So I'm reluctant to spend time on this unless I receive an encoder to test myself.

keirf, 2018-06-13:
Okay could you please try the following firmware, requiring `rotary = half` in FF.CFG:

[ff_130_1.zip](https://github.com/keirf/FlashFloppy/files/2098635/ff_130_1.zip)

TzOk83, 2018-06-13:
Works perfect!

keirf, 2018-06-13:
That's great! I will test on my basic full-cycle encoder tomorrow (I changed round the whole rotary encoder logic to make it more straightforward and logical). If it works okay I will commit the patch to master and it will get rolled into next release 0.9.22a.

closed 2018-06-14 by keirf

tkurbad, 2018-06-17:
@keirf Thanks for implementing this. I bought another encoder meanwhile but it does not fit my 3d printed external case so well as the 'half' cycle encoder I designed it for. I just thought, I'd have to live with that. It's nice to see that I can put the original encoder back now!
By the way, the new encoder also has a green base, but did just work with the 'simple' setting. So, there seems to be a lot of scattering in this area...

commit b3ac2acc (2018-06-14): rotenc: Support half-cycle-per-detent rotary encoders
Rationalise FF.CFG rotary option to {none, quarter, half, full}.

Fixes #130