keirf/flashfloppy #132: Commodore 64 1581 .d81 disk image support?

fzdravko, 2018-06-18:
Any chance for Commodore 1581 .d81 disk image support?

Thanks.

keirf, 2018-06-18:
Tentative yes but depends as I thought 1581 had a custom serial interface to the c64 and the Gotek would not really be physically compatible?

fzdravko, 2018-06-18:
Yes, it has custom serial interface for connection with c64, but internally 1581 have plain 3.5" floppy drive and connection to it (like amiga...). 

P.S.
I also have usb HxC Floppy emulator and it is working with 1581.

keirf, 2018-06-18:
Oh I'm thinking of 1541 with custom formats and different internal drive interface. Do you have documentation for the d81 image and physical track format?

fzdravko, 2018-06-18:
No, I don't, but I found this
https://ist.uwaterloo.ca/~schepers/formats/D81.TXT

Also on HxC forum there is a mention that D81 in nothing special:
http://torlus.com/floppy/forum/viewtopic.php?t=531

keirf, 2018-06-20:
Okay please try the following firmware:

[ff_132_1.zip](https://github.com/keirf/FlashFloppy/files/2118219/ff_132_1.zip)

fzdravko, 2018-06-20:
I tried it but it doesn't work. Just says "file not found".

tomse, 2018-06-20:
Now that the issue is up, I guess the GOTEK isn't powerful enough to actually act as a 1581/1541/1571 drive which can directly be attached to the IEC port.?
instead of having a shugart/pc interface port with the 34pin connector, i.e. pins 1-5 can be used as the serial/iec interface instead.

Squirrel61, 2018-06-20:
@tomse I guess you're talking about sd2iec or similar solutions? In other words, it has already been made. So it's perfectly doable, but a waste of @keirf 's precious time. And besides, I think it would be an "either-or" solution, because I guess it will be impossible to have the Gotek support both the IEC bus and the normal floppydrive bus at the same time. I'm already wondering with al the additions @keirf is making, when the Gotek will be out of flash memory.

tomse, 2018-06-20:
@Squirrel61 yes similar to the sd2iec, but where sd2iec isn't 100% compatible, far from it, a cheap and better solution could lie in a gotek. Yes there is also 1541 Ultimate which is close to 100% compatible this one doesn't support d81 files though.

If it's a waste of Keir's time only he can judge that.

fzdravko, 2018-06-20:
@keirf Yes, it's connected right.

keirf, 2018-06-20:
And jumpered right? I don't know whether you need a jumper at JC (selects PC style interface rather than Shugart). Also S0 vs S1 for drive select. Do you see the green Gotek led flash on drive activity? Do you see the track number change as the drive heads are driven by the host?

fzdravko, 2018-06-20:
I didn't change any jumpers, will try that later. When 1581 and gotek are on and image file selected, when I restart c64 (turn off/ turn on) tracks go to 0 and then to 39:1. When command for dir is issued, error "file not found" is immediately shown, it doesn't even try to read.

I will try with different jumper config later today.

keirf, 2018-06-20:
Actually I suspect the jumper configuration is fine, interface is Shugart which is the default (no jumper at JC). Also probably drive-select 0 is correct, so jumper at S0 only. I guess this is what you already have. Now I am suspecting the track layout within the image is different than what I expect. Let me check that,,,

pdp11user, 2018-06-20:
The 
[image]
Commodore 1581 controller needs ready at pin 34 and drive-select  0:

interface =shugart / jc = open
DS 0

Hope that helps!

keirf, 2018-06-20:
Okay, looks like side ordering is flipped in a D81 image. Here is a new firmware to test:

[ff_132_2.zip](https://github.com/keirf/FlashFloppy/files/2119476/ff_132_2.zip)

fzdravko, 2018-06-20:
It's working now.

Thank you!!!

closed 2018-06-20 by keirf

gilmod714, 2024-12-16:
it would be really nice to have FF connect directly to IEC bus .. here's a scenario: you have VICE emulator running on the PC and zoom-floppy USB adapter (IEC) adapter .. you want to attach physical floppy drives (any shugrat/PC type) 3.5 / 5.25  and run original c64 disks .. btw : vice can connect to a real c64 drive using zoom-floppy/or DIY promicro clone

commit a82307fc (2018-06-20): img: Fix D81 image sode ordering.

Fixes #132
commit 5f1b52ae (2018-06-20): img: Fix D81 image side ordering.

Fixes #132