keirf/flashfloppy #133: Support request for Ohio Scientific disk formats

tenuke, 2018-06-18:
(Philip Lord had posted this in facebook so I thought that I could aswell copy this into here myself too)
([link to post](https://www.facebook.com/groups/flashfloppy/permalink/947445588749008/))

Hi Keir,
Would it be possible to have flashfloppy support Ohio Scientific disk formats?

The specs are as follows (thanks to Dave)

There are two different bit rates. For 5.25" disks, the data rate is 125 kHz. An FM circuit produces a clock pulse, and a data pulse (if 1) for every bit. The data is written with a UART (start bit, 8 data bits, even parity, 1 stop bit). For 8" disks, it's 250 kHz.

Track format (except track 0, see below) is:
- (INDEX DETECTED)
- 1ms delay after index
- 0x43, 0x57 /* track header start */
- Track number /* one byte */
- 0x58 /* track header stop */
- nominal 6.6 us delay after header / * may vary with system speed */
Sector format: 
- 0x76 /* sector header start */
- sector number /* one byte */ 
- pages in sector */ one byte */
- sector data /* #pages x 256 bytes */
- 0x47,0x53 /* sector end */

Delay between sectors, nominally 800 us for each page in previous sector. In practice, this varies.

Track 0 is usually reserved for boot code, and has a different format, due to the simplicity of the bootloader in ROM. There is no sector. The header is just the load address and number of pages to load, followed by the data:

- INDEX PULSE
- 1 ms delay
- high byte of load address
- low byte of load address
- # pages /* one byte */
- data

So, to clarify, this is a simple FM format.

The bitrate is 125 kbps for 5.25" drives (pulse rate up to 250 kHz including clock and data pulses)

The bitrate is 250 kbps for 8" drives (pulse rate upt to 500 kHz including clock and data pulses.)

keirf, 2018-06-18:
This one should probably work already as HFE.

v-retro, 2018-06-22:
Hello,

I was wondering if it would be possible support Atari ST’s MSA (and the
compressed version as well) please?

It’s a far more efficient format (storage wise) and appears to be in wide
usage as well

http://fileformats.archiveteam.org/wiki/MSA_(Magic_Shadow_Archiver)

Thanks
Paul
-- 

*Paul Vatta+61409130512*

keirf, 2018-06-22:
MSA is not relevant to this ticket. However I have no plans to support compressed formats as they are rarely convenient for streamed reads and writes.

keirf, 2018-10-30:
Won't support this one direct in firmware. Conversion HFE<->image files would be nice but I think there is already work in the OSI community to support this.

closed 2018-10-30 by keirf