keirf/flashfloppy #136: ERROR : HxC drive not detected

Duduche87, 2018-07-08:
Hi,

A600, CF4GB, Gotek FF 0.9.22 (Buzzer on JB, Jumper on S0)

So Amiga Autoboot.hfe give me : ERROR : HxC drive not detected !!!
No FF.CFG file on USB Key, with I have nothing !

With another usb key used with another FF Gotek, the game on slot 001 load normaly, so my Gotek is OK no ? Same error on slot 000.

So what is the problem ?

Trying to update : E34 !

Thanks for your help.

keirf, 2018-07-08:
What version is the autoboot file (displayed on bottom row of screen when it starts up)?

Can you put just one ADF file (Arte demo) nothing else on the USB stick and tell me if that runs okay?

E34 is not an error of the update process so you have failed to enter the updater. Make sure both buttons are pressed at power on and you should see UPD on the Gotek led display.

Duduche87, 2018-07-08:
Now I am on 0.9.23a (all flashed). Autoboot.hfe ver 1.71

With official HxC 1.65 no problem but message that firmware is not official.
So I appreciate a fonctionnal FF version.

Thanks.

keirf, 2018-07-08:
Please try Arte ADF as the only file on USB stick, as I previously asked.

Duduche87, 2018-07-08:
I have renamed a game in Arte.adf, alone on usb key, the game boot normaly.

Duduche87, 2018-07-08:
Why this specialy ? For the trackdisk fonctions ? I download it, but the game work fine. So flash is OK.

keirf, 2018-07-08:
Yes it is known to use the ready disk signal (pin 34) same as FF file selector uses for detecting a Gotek. So if the demo boots and runs for you, that aspect of the hardware must be OK...

keirf, 2018-07-08:
And if it doesn't work, the problem is in the Gotek or the amiga or (not unlikely) the cable

Duduche87, 2018-07-08:
Ok, so Arte doesn't work. So jumper on M0 ? Because with HxC on Amiga non jumper on M0 only on ATARI and on Amstrad M0 is now for the drive A or B.

keirf, 2018-07-08:
Do you have another cable you can try? Or Gotek? Or Amiga? This way you can narrow down the item at fault.

keirf, 2018-07-08:
Your jumper configuration is already correct for Amiga.

Duduche87, 2018-07-08:
The cable was the cable with my second A1200 (with now a dual cable for gotek cortek and floppy drive inside), this A600 was received without drive.
All is closed. I will see for another cable but for a gotek test I must open the 600.

keirf, 2018-07-08:
The reason the cable can be suspected is because ribbon cables get stressed by disconnecting and reconnecting, and tend to rupture at an outermost wire where it is punctured by the IDC connector. Ready signal is pin 34 which happens to be one of those outermost wires. It is not rare at all. And of course if it is this, it's good news as cheapest thing to replace.

Duduche87, 2018-07-08:
Works fine with autoboot 1.70 on my 1200 DF1 !

keirf, 2018-07-08:
1.71 needs more recent FlashFloppy firmware version. Please update your 1200 FlashFloppy to latest version and try 1.71 again.

Duduche87, 2018-07-08:
Error on 600 again so I made another cable.

Duduche87, 2018-07-08:
OK thanks. It was the cable. So I made on with the connectors in good way.
Big thanks. I close the 600.

closed 2018-07-08 by keirf