keirf/flashfloppy #137: Ensoniq EPS classic  "Disk has been changed" in shugart mode, or no disk change in ibmpc mode

evx73, 2018-07-08:
Hi there -

I tested FF with an Ensoniq EPS classic. The OS loads in both shugart mode and ibmpc mode.

When in shugart mode, I experience a behavior similar to what is described here http://www.torlus.com/floppy/forum/viewtopic.php?f=17&t=510&start=15

It seems that the OS reads the FAT at each file access. If I switch to ibmpc mode, I can read the disk images with not issue, but the OS fails to detect disk changes.

The original floppy is a Panasonic JU253 13P 
https://ia800308.us.archive.org/28/items/bitsavers_panasonicfchFloppyJU253JU257Brochure_1972274/Panasonic_3.5_Inch_Floppy_JU-253_JU-257_Brochure.pdf

Can you recommend any further test?

Thanks!

keirf, 2018-07-08:
No immediate great ideas. Perhaps try the other interface= options in FF.CFG and report how they perform?

keirf, 2018-07-08:
Another possibility is a bad floppy cable.

keirf, 2018-07-11:
Some Ensoniq definitely require `interface=ibmpc-hdout` in FF.CFG. So you should definitely try that. I think it will fix it. Please reopen if not.

keirf, 2018-07-16:
Seems others have this same problem. Reopening.

evx73, 2018-07-16:
Hi - I am currently away, but I will test the Gotek/EPS combination again when I am back. I did try all "interface" possible values via FF.CFG. None of them fully worked.

keirf, 2018-07-17:
Attached here a firmware to test which allows greater configurability of pins 2 and 34.

[ff_137_1.zip](https://github.com/keirf/FlashFloppy/files/2201743/ff_137_1.zip)

In FF.CFG:
```
pin02 = ...
pin34 = ...
```
where `...`:
```
auto # Automatically assigned based on interface=
nc    # Unused / No Connection
high # 5 volts
low  # 0 volts
rdy  # Ready
nrdy # Inverse of Ready
dens # Density (HD = 0v)
ndens # Inverse of Density
chg # Disk Change
nchg # Inverse of Disk Change
```

keirf, 2018-07-18:
Had a report of success with:
```
interface = shugart
pin02 = nchg
```

thestudioman, 2018-12-17:
EPS Classic Working With 
image-on-startup = static (Specify Operating System EPS OS v2.49.img to Load at Startup in IMAGE_A.CFG
autoselect-file-secs = 1
interface = shugart
host = ensoniq
pin02 = nchg
pin34 = chg

Extract Floppy with Omniflop to Floppy Images Format.img 
Copy Operating System at Root of Thumb Drive
EPS OSv2.49.img

thestudioman, 2018-12-17:
EPS Classic Working With
image-on-startup = static (Specify Operating System EPS OS v2.49.img to Load at Startup in IMAGE_A.CFG
autoselect-file-secs = 1
interface = shugart
host = ensoniq
pin02 = nchg
pin34 = chg

Extract Floppy with Omniflop to Floppy Images Format.img
Copy Operating System at Root of Thumb Drive
EPS OS v2.49.img

thestudioman, 2018-12-17:
EPS Classic Working With
image-on-startup = static (Specify Operating System EPS OS v2.49.img to Load at Startup in IMAGE_A.CFG
autoselect-file-secs = 1
interface = shugart
host = ensoniq
pin02 = nchg
pin34 = chg

Extract Floppy with Omniflop to Floppy Images Format.img
Copy Operating System at Root of Thumb Drive
EPS OS v2.49.img

On a Gotek Drive if you want to Load a Second Instrument From the Same Floppy you have to Select Another Floppy img on the Gotek and Then Go Back to the Desired Floppy Before Pressing Load ! once load is Pressed Change to the Desired Instruments with the up or down Arrows on the EPS then Click Enter and Select Button for Instrument
if you Ever See "Disk Drive Not Ready" you have to Select Another Floppy img on the Gotek and Then Go Back to the Desired Floppy Before Pressing Load !

thestudioman, 2018-12-18:
EPS Classic Working Perfectly With
image-on-startup = static (Specify Operating System EPS OS v2.49.img to Load at Startup in IMAGE_A.CFG
autoselect-file-secs = 1
interface = shugart
host = ensoniq
pin02 = nrdy
pin34 = rdy

Extract Floppy with Omniflop to Floppy Images Format.img
Copy Operating System at Root of Thumb Drive
EPS OS v2.49.img

if you Ever See "Disk Drive Not Ready" you have to Select Another Floppy img on the Gotek and Then Go Back to the Desired Floppy Before Pressing Load !