keirf/flashfloppy #138: Extended DSK don't work in v0.9.23a

sirdavid2, 2018-07-08:
I have updated two Goteks to v0.9.23a and since then none of EDSK images works. CPC tries to read something and it reads with errors. SAM Coupe can't read anything and the virtual head moves like crazy. Other file formats on SAM work ok. With 0.9.22a all worked ok.
[edsk.zip](https://github.com/keirf/FlashFloppy/files/2174058/edsk.zip)

keirf, 2018-07-08:
That's a pair of CPC EDSK images right? I'll reproduce this issue on my CPC and report back.

sirdavid2, 2018-07-08:
No. One is CPC (Batman) and one SAM.

keirf, 2018-07-08:
Oh of course, now I recall seeing the BTTP demo before. Sadly I don't have a Sam but I will test Batman disk on CPC, fw 9.22 vs 9.23.

TomDDG, 2018-07-09:
Just to let you know I'm having a similar issue with ZX Spectrum eDSK. Some of the ones I tested in 9.19a now don't work in 9.23a. Reverting firmware back to 9.19a and they work again. Tried clearing cache and re-flash but doesn't fix it. An example of this is Xybots from the TOSEC collection. Can send the disk if you need.