keirf/flashfloppy #140: Protected ZX DSK images

merlinkv, 2018-07-11:
I updated my Goteks to 0.9.24a but still can't run protectected dsk images like OutRun (1989).

keirf, 2018-07-11:
See this PDF document http://tomdalby.com/other/plus3gotekcompat.pdf
Some titles (like Out Run) have incomplete copy-protection info in the TOSEC DSK image. Luckily the required info can be added automatically. There is a Python script in the FlashFloppy repository. You could run it something like this:
```
# git clone https://github.com/keirf/FlashFloppy
# python FlashFloppy/scripts/edsk_fix_speedlock.py OutRun.dsk OutRun_Fixed.dsk
```
Does that make sense?

keirf, 2018-07-11:
You probably should invoke directly from the cmd prompt like this:
```
python edsk_fix_speedlock.py OutrunA.dsk OutrunA_Fixed.dsk
```
Also, the script needs Python2, so you need to install Python 2.7.14.

merlinkv, 2018-07-11:
Hi keirf.

Thanks a lot.

I will try with Python 2.7.14 ....

merlinkv, 2018-07-11:
Hi keirf.

With Python 2.7.14 it ..... Works !!!!!!!!!!!!!

Thanks you very much.

keirf, 2018-07-11:
Thanks I will document this in the Wiki and then close the ticket.

I will also probably convert all scripts to Python3. It's quite prevalent now, I should make the switch.

merlinkv, 2018-07-11:
Sorry but works only on side A ...

Checking the Outrun Side B ... don't recognize as Speedlock.

Opening the image in ZX-BlockEditor it seems like is also Speedlock protected ....

keirf, 2018-07-11:
Agreed. Let me fix the script....

keirf, 2018-07-11:
Try again with the latest script. I just removed the signature check as clearly not all protected disks/sides have the signature.

merlinkv, 2018-07-11:
Side B still the same .....

On +3 menu, pressing Loader, starts to load, but stops on 01.0

If I try with LOAD "*" .... Disk not bootable ....

keirf, 2018-07-11:
Is Outrun Side B actually bootable? The first sector is empty so I would guess not. It's just track data probably? Try the Side B image in an emulator.

TomDDG, 2018-07-11:
I've just checked my re-dump of the original disks and can confirm Side B doesn't have the SPEEDLOCK header as you mention but does have the extra information in track 0 sector 2 for weak sector protection. Side B doesn't boot as it is just additional track data, however the first 3 stages are on Side A so you have to get a lot further in the game for it to be an issue.

keirf, 2018-07-13:
NB. I have updated all Python scripts to be compatible with Python 3 *and* Python 2.7.

TomDDG, 2018-07-14:
Just as an FYI, and unbeknown to me, the SamDisk software (http://simonowen.com/samdisk/) which is used to image 3" disks can also be used to patch the disk images so that they work, such as adding the weak sector information. You basically run "SamDisk image.dsk fix.dsk --fix". I've tested it and it works fine, I think it also works with other protected disks in particular those on the Amstrad where the weak sector maybe not on track 0.

keirf, 2018-07-14:
I'll confirm and document it in the wiki and remove our script, if so.

keirf, 2018-07-18:
Updated the Wiki to include instructions on both methods.

closed 2018-07-18 by keirf