keirf/flashfloppy #141: dsk dragon32/64 - COCO dsk/VDK

retrocant, 2018-07-12:
Greetings, could you implement the flashfloppy to support the discs of Dragon 32/64 and coco tandy?

http://archive.worldofdragon.org/index.php?title=Tape%5CDisk_Preservation

retrocant, 2018-07-12:
I just saw a VDK file in a hexadecimal program and it is not compressed.

retrocant, 2018-07-12:
We have a disk image that calls HOBBIT1.DSK or HOBBIT1.VDK, so that it can be correctly converted to HFE (for HxC) it has to be renamed to HOBBIT1.JVC. Once renamed it can be converted without problems and voila! -thumbup

If it is not renamed to .JVC the resulting .HFE image ... WILL NOT WORK !!!

keirf, 2018-07-12:
Feel free to attach the original hobbit image here too.

retrocant, 2018-07-12:
ok, i will try

retrocant, 2018-07-12:
Greetings, it is better that users, if possible, do not have to change the extension, if you can make it internally change the extension much better.

retrocant, 2018-07-18:
greetings keirf, how is it going with the implementation of the dragon / tandy part?

keirf, 2018-07-18:
Not implemented yet but high on the todo. I will try to get it done by the weekend.

keirf, 2018-07-20:
Here is a test firmware to try. To use images with suffix .dsk you need the following in FF.CFG:
```
host = dragon
```

[ff_141_1.zip](https://github.com/keirf/FlashFloppy/files/2213977/ff_141_1.zip)

retrocant, 2018-07-20:
Greetings, I just tried it and it gives me flaw in reading with the flagonbird.

in post host = dragon.

I have changed the extension from VDK to DSK.

retrocant, 2018-07-20:
Greetings, it works.

I have changed the file to .vdk to its original extension and if it works.

keirf, 2018-07-20:
That's great and yes, to clarify, you do not need to change file extensions.

keirf, 2018-07-20:
Please also try a jvc/dsk or two.

retrocant, 2018-07-20:
Greetings, I do not currently have the tandy / coco controller to be able to test the dsk.

keirf, 2018-07-20:
Is jvc/dsk particular to Tandy rather than Dragon then? The only thing that host=dragon does is allow dsk file extension to be used for jvc images. If they are not really used for Dragon then perhaps host=tandy would be better?

retrocant, 2018-07-20:
Greetings, the head of the dragon / tandy discs is different, what you say about generating host = dragon for the vdk

host = tandy for the dsk

I hope people can test for Tandy CoCo

keirf, 2018-07-23:
Okay this is committed to master for inclusion in forthcoming v0.9.26a. If you wish to test beforehand then here is a new test firmware. If you wish to use DSK images then you now need `host = tandy-coco` in FF.CFG.

Another note: sector interleave on VDK images is 2:1 (suitable for Dragon DOS). Sector interleave on JVC/DSK images is 3:1 (suitable for RSDOS). Assumption is that Coco images (JVC/DSK) would only be used with Radio Shack's DOS. If the interleave would better be dependent on `host = tandy-coco`, rather than on image type, just let me know. It's easily changed.

[ff_141_2.zip](https://github.com/keirf/FlashFloppy/files/2218474/ff_141_2.zip)

closed 2018-07-24 by keirf