keirf/flashfloppy #142: Feature suggestion - Auto mount/eject

RetroClinic, 2018-07-13:
Hi. A suggestion or three, if it's possible?

How about instead of pressing the encoder to load a file, then again to eject, or have the constant access to the USB stick, have an option for an auto-insert once the drive is selected, and another one for an auto-deselect (eject) after a set period.

I'm not sure how it works in other systems, as I only use it with the BBC Micro, but it would be a way of automatically ejecting a disk to prevent the stick corrupting if turned off when it's being access constantly.

Would it also be possible to have another option that locks the disk image when the current file is selected? So it's not possible to use the encoder or buttons to change image, while the machine can read/write from the selected one.

Thanks, Mark.

PS great fix on the BBC sector skew! Makes them perform so much better.

TzOk83, 2018-07-13:
Inserting/ejecting disk is only virtual, it has nothing to do with the access to the stick. Drive is accessed only when host computer requests read/write access to the floppy. In most systems a feature of ejecting a floppy (image) is obsolete.

keirf, 2018-07-13:
These are kind of valid requests. There seems to be some claim that constant USB stick reading can cause failures, though that doesnt really tally with the theory of operation of flash. However that will be fixed when I implement a buffer cache. Locking against image selection while drive is selected... I could make that a config option perhaps.

closed 2018-10-30 by keirf