keirf/flashfloppy #145: TR-DOS SCL image format

xolod79, 2018-07-27:
Please add support SCL (ZX-Spectrum TR-DOS) image format. 
Description of the format
Headline:
+0, 8 bytes - SINCLAIR signature
+8, 1 byte - number of files
Data:
Headers of files (14 bytes from the directory without track / sec):
Title of the 1st file
Heading of the 2nd and subsequent files (if any)
File data:
Code block of the 1st file
Code block of the 2nd and subsequent files (if any)
The last 4 bytes are the control sum (the arithmetic sum of all previous bytes).
HxCFloppyEmulator_soft has the support of converting this format to HFE format.

keirf, 2018-08-07:
Not really as this is not a block image format. Can you convert SCL to TRD using HxC software?

Alternatively I would need to do this conversion dynamically in firmware. Unlikely to happen.

xolod79, 2018-08-10:
Yes hxc software converts scl to frd. the scl format is very similar to trd and differs only in the first track which contains a list of files in which there is no only CHR location on the disks. and information about disks (9 sector).

Volutar, 2018-08-30:
Non block images such as SCL still can be used as readonly images with ease. And SCL images actually are not meant to be writable. 

Maybe it's not bad idea to support some formats in readonly mode?

xolod79, 2018-08-30:
I agree with you, readonly is enough.

Volutar, 2019-07-17:
The main "issue" with SCL, is the absence of track/sector location for each file, and absence of "disk specification" sector (0:8). The length in sectors is enough, to calculate "original" actual file location. Since TRDOS file data is always starting with track1 sec0 (it is the rule), so the rest can be filled in on the fly. Disk type is always treated as 80 tracks 2 sides, but actual track number that SCL can give access to is not limited with 84 (so it can pass track number 210, which exceeds TRD format capability, and some images use that). Disk label can be taken from the file name, and first free track/sector also can be calculated.

Adding extra file to SCL image would require catalog area expanding, to fit extra 14 bytes for the file header, and CRC recalculating after adding actual file data to the end. So, generally, it would require whole image file rewrite (which is impractical). Thus it would be much more effective to add forced "readonly" mode for certain image types (SCL in this case). By the way, it would also make easier introducing of new disk image formats in "native" mode (simply by skipping writing mechanic - in most of cases readonly mode is enough).

SCL specs can be taken from here: http://www.zx-modules.de/fileformats/sclformat.html
TRD (for comparison) is here: http://www.zx-modules.de/fileformats/trdformat.html