keirf/flashfloppy #149: MB02 support for ZX Spectrum

engcameraman, 2018-08-03:
Could you support the MB02 floppy format for ZX Spectrum as well please?
Here's what I've found:
The disk fileformat is .MBD (a plain dump of an MB-02 floppy, in a fashion similar to .MGT), typically high-density 82 tracks x 11 sectors of 1024 bytes each

keirf, 2018-08-07:
Should be easy. I will investigate.

keirf, 2018-08-13:
Please try the attached firmware with your MBD images. As well as working, interested to know if read/write speed is approximately correct, too.

[ff_149_1.zip](https://github.com/keirf/FlashFloppy/files/2283179/ff_149_1.zip)

engcameraman, 2018-08-20:
I've been trying tonight, I'm getting no response from the Gotek on my MB02, I don't know what jumper settings to use. I've tried all kinds of settings. Does the image need a special extension name other than .img?

keirf, 2018-08-20:
The extension needs to be MBD. I thought that was normal for these images? If you want to use IMG extension it will need me to add a FF.CFG option.

engcameraman, 2018-08-20:
Yes MBD is normal, but not thinking it through, I had put them on the flashdrive before your new firmware was made and renamed them as IMG when I first tried the Gotek on the MB02.
I’ll rename and give it another try soon,

Thanks,

Frans van Egmond
frans@playy.nl

engcameraman, 2018-08-21:
The MB02 boots fine from an imagefile now, and looks to be at the same speed as a real floppy.
My keyboard on the Spectrum seized working so at the moment I cannot test further...
Thanks!

commit 8c18969b (2018-08-21): img: Support *.MBD image files for Spectrum MB02
Fixes #149
closed 2018-08-21 by keirf