keirf/flashfloppy #152: UFA1M44?

sandgoose, 2018-08-08:
browsing around to see if i can find something local and run into several UFA1M44 "Normal version" listed.

pictures show, description states "34pin floppy driver interface" but also something about jumpers for supporting different hardware...

mostly wondering because these are local, anyone tested these ? worth trying or no?

keirf, 2018-08-09:
Any Gotek with 34 pin floppy header will work. But be careful here because UFA1M44 is the model number which has a mini USB port at the back instead of floppy header. This is not what you want!

Since the listing is clearly incorrect one way or the other, you can't trust it. Maybe contact the seller.

sandgoose, 2018-08-09:
thank you for the response, I have maybe a more interesting question!

would it be possible (via tricks and I guess amiga hardware controller permitting) to emulate a floppy disk larger than a HD 1.76MB amiga formatted floppy as as sort of a pseudo/poor mans hard drive for older machines lacking ide!?

Thanks

keirf, 2018-08-09:
Yes it would require a new device driver on the amiga host though. It is something I'd like to pursue at some point (actually direct access to the USB stick itself)

closed 2018-08-09 by keirf