keirf/flashfloppy #153: Unauthorized disk ejection when writing 100k+ files

ChristopheVC, 2018-08-09:
Hi,
when using FlashFloppy on an industrial equipment, everything goes well with small files. (reading & writing)
when the system tries to write large files, the system stops each time at the same %, saying there was an unauthorized ejection of the disk. It must be at around 100kB of the file.
Also formatting does not work and stops after a certain % of the total process. Again, each time the same %
I tried with img and ima files. 
Also tried different combinations in the config file. (interface, host, side-select-glitch-filter, track-change, index-suppression)

Any ideas? 
Thanks

ChristopheVC, 2018-08-10:
I use 1.44M img files.
The system says that it has 1457664 of free space, the same amount as with floppy disks.
I made the img files with 2 different programs, but with the same result.
I used another type of USB stick, also with the same result.

If i look at the Gotek busy light, then it goes out a couple of times, but the system continues untill that point that the system says the disk was changed/ejected. 
I took a look at the file on my floppy backup and the systems stops after writing +/- 70000 bytes

I have no idea of the disk layout which is used by the host, I will try to figure that out. With floppy disks the host accepts windows formatted floppies without any problem.

keirf, 2018-08-10:
Do you have to format windows formatted floppy disks before you can use them?

ChristopheVC, 2018-08-10:
I can use both, pre-formatted and "re"-formatted floppy disks. Reformatted with a windows machine or with the host.

keirf, 2018-08-10:
Then try another USB stick, clear the flash configuration (hold both buttons with no stick inserted until display changes) and try `index-suppression = no` as the only line in FF.CFG.

Interesting if this fixes the issue or even just changes the progress percentage at which write or format fails.

ChristopheVC, 2018-08-10:
Now I get a "no disk" error

keirf, 2018-08-10:
Have you tried a jumper at position JC at rear of the Gotek? Or same effect: `interface = ibmpc` in FF.CFG.

ChristopheVC, 2018-08-10:
The progress % stays the same.

keirf, 2018-08-10:
Does this system work with unmodified PC floppy drives do you know?

keirf, 2018-08-10:
Also I'd expect changing interface to make some difference. I wonder if your FF.CFG is being picked up at all?

ChristopheVC, 2018-08-10:
When I use no JC jumper, he seems to recognize the interface = ibmpc parameter

ChristopheVC, 2018-08-10:
What do you mean with unmodified PC floppy drives? When I insert a brand new IBM formatted high density floppy drive, then it is recognized and usable. I do not have any unformatted floppy drives at my disposal for the moment.

keirf, 2018-08-10:
I am talking about the drive, not the disks.

ChristopheVC, 2018-08-10:
For the moment we are using TEAC fd-235hf floppy drives. So that are PC compatible drives.

keirf, 2018-08-10:
So I would try jumpers on S1 and JC only. No FF.CFG file at all. And reset flash configuration by holding both buttons with no stick inserted until the display changes (led display changes to rSt).

This is a totally basic setup for FlashFloppy in PC compatible mode.

keirf, 2018-08-10:
Could you try `interface = ibmpc-hdout` in FF.CFG? You tried more than one USB stick now with zero effect, right?

ChristopheVC, 2018-08-10:
Now there is change.
The random stops are gone. The system stops always at 30% even after a flash configuration reset.

I tried more than one brand of USB stick with the same results.

keirf, 2018-08-10:
I don't know it seems perhaps some nonstandard configuration is needed for pins 2 and 34. Some of the teac drives do have jumpers for configuration so it's not a given that the drive is set up just as it would be for PC. Do you have any documentation on the host system or procedures for changing the drive?

ChristopheVC, 2018-08-10:
I will try some configurations of pin 02 & pin 34.
As far as I've tested yet, it seems to me like the voltage on pin 34 drops to 0V  for a little while at 30% writing.

I do not have detailed documentation on the drives of the host.

keirf, 2018-08-10:
Pin 34 going low would be unexpected and bad for a PC drive as that is the dskchg signal and should only be low when no disk is inserted.

Perhaps try `pin34 = high` in FF.CFG. This would stop disk changes from working but is okay for your basic write tests I think. If pin 34 continues to go low then this signal has unexpected purpose, or there could be something electrically wrong with the host or the Gotek.

ChristopheVC, 2018-08-10:
The host does not allow pin 34 to be high all the time. 
I noticed something else. When I insert a USB without IMAGE_A.CFG, the fault appears allmost immediatly when the host starts writing. (probably with the creation of the file)

keirf, 2018-08-10:
Very odd behaviour if you can be sure that pin 34 changes value during writes then something is definitely up.

keirf, 2018-08-13:
Let me know if there's more advice I can give here. Else I will close this ticket, as I'm not sure there's a bug fix or documentation-update to be gleaned here.

ChristopheVC, 2018-08-13:
For the moment it's ok. I will do some further testing on my own.
Thanks for the advice.

ChristopheVC, 2018-08-13:
I did some further reading on the Gotek and i think it could have something to do with the memory of the Gotek being 64kB. I guess, when the memory is full, the Gotek starts writing to the USB and looses connection with the host.

keirf, 2018-08-13:
Certainly the Gotek can go quiet while it is writing back to USB but most hosts can deal with this okay. If it was an issue of writeback speed, you might expect changing USB stick to alter behaviour (eg % progress before failure). So it still feels like something else is afoot here.

keirf, 2018-08-13:
I think short of getting a debug build running to collect serial logs, or getting direct access to this system myself, we'll have to park this for now. If you wish to pursue this further, or can collect more info (eg. interesting info on floppy interface signals such as pin 34 via scope or logic analyser) then feel free to reopen.

closed 2018-08-13 by keirf

ChristopheVC, 2018-08-16:
Hi,
I just want to share my solution.
All the USB sticks that i have tried were USB2.0. So I bought a USB3.0 stick with fast writing speed and now it finishes writing as it should.
So after all, it was due to the too slow writing speed to the USB stick.

keirf, 2018-08-16:
Glad to hear you're sorted. Hopefully I can get performance better for a wider range of sticks post-v1.0.