keirf/flashfloppy #155: Info: FF with 26-pin header Gotek Floppy

JoanMCD, 2018-08-11:
I sucessful use FF with an 26-pin header Gotek Floppy and use it in my Tektronix TDS3000 series (TDS3012B) Oscilloscope. The PB and the LEDs are at the same pins. I have to remove the resistors RA & RC (normaly 0R), and (not sure if it's nessesary) I also removed RB and the external EEPROM (8pin SOC). The PCB No. of the GOtek is SFRC2D.

Thx for the great project!

keirf, 2018-08-13:
Do you know what RA, RB, RC (all 0R?) do or how they are connected into the Gotek schematic, and why you had to remove them? If I can document that I can add something to the "Gotek Compatibility" section of the wiki.

JoanMCD, 2018-08-13:
Yes, all are assembled with 0R. With RA the drive shows 'Upd' on the display. The drive is mounted now, I cann't measure the Rx->pin config. But i have a fast drawing on paper - maybe I can re-read it from there. 

From [here ](http://torlus.com/floppy/forum/viewtopic.php?t=2019) we can know that RA is pulling down Pin 37 of the STM32
Over all, the info is that by removing the resistors the 26-pin drive is also useable.

keirf, 2018-08-13:
Added a link to this ticket to the wiki.

closed 2018-08-13 by keirf

JoanMCD, 2018-08-13:
[image]

[image]

presslab-us, 2019-06-10:
Thanks for the info, I was able to get the SFR1M44-DU26 working on my LeCroy LC684 oscilloscope.  I only removed the three resistors and left the EEPROM.

mbirth, 2019-09-16:
Did it not work before? I have the SFR1M44-DU26 happily working in my TDS784D without any modifications. Just curious as to why you guys modded your disk emus.

oldtrusty, 2019-12-31:
I also have the SFR1M44-DU26 26 pin version and having trouble getting the oscope to recognize my USB stick.  Currently have the RA, RB, & RC 0R resistors removed.  Only jumper in use is the S1 jumper on the rear header.  I have successfully upgraded the firmware to FF and tried v3.9a and 2.13 with the same results.  My scope constantly gives me a message to insert floppy disk, which leads me to believe that my USB stick is not properly formatted.  You can see in the pic that my oscope finds the drive ok and lists the directory screen.  It just won't read my files (previous oscope screen captures while using the floppy drive) that I placed on the disk for testing.  I have tried a 1Gb and 2Gb stick formatted into (99) 1.4Mb disks.  Looking for some assistance in how others who successfully got this working and how you all prepared your USB drives and what image settings you used.  To this point I have tried every option in FF.CFG (placed in the root directory) with no success.  Also tried removing serial eprom - no difference.  The USB drive does however increment/decrement the virtual floppy disks when I press the front up/down buttons.  I can see the LED inside of the USB drive and the ones on the emulator give feedback when I try to access the drive.  So it seems to be somewhat working.  See attached photos.  Any help on this would be greatly appreciated.
[image]
[image]
[image]
[image]

mbirth, 2019-12-31:
IMHO the original firmware of the SFR1M44-DU26 works better. With no USB device connected, it uses the internal flash memory as a virtual disk. Once you've connected a USB device (one large partition formatted to FAT32) and press the right button, it'll then read the first few files (up to 1.44 MB) from the USB and present those as the "disk". Pressing the button again will read the next few files and so on.

And if you want to store files on USB, just press the left button and from that moment on, all files written to the "disk" will end up on the USB drive. No need to make multiple disk images or special partitions or anything.

JoanMCD, 2019-12-31:
The Tektronix TDS3000 series have a 26pin flat wire, so to use the SFR1M44-DU26 is a logic step. The original FW offers only a non HD Disk - what was not accepted by (my) TDS, so to replace the FW to get a HD drive was the next logical step.

JoanMCD, 2019-12-31:
@oldtrusty  I see a lot of markings on your PCB, and also a TO92 Transistor - ... As i remember, at the end I didn't need such modifications.

oldtrusty, 2019-12-31:
@mbirth do you have the original firmware or know of a link to it?

oldtrusty, 2019-12-31:
@JoanMCD the markings were to help me identify the boot0, +3.3, Tx, Rx, and Gnd pin during programming of FF.  The TO92 (2N2222A) is temporary and was to replace the 1P SOT23 (2N2222A), which I damaged when I was playing with the rear jumpers.  It tells the uC that the disk is read-only, I believe.  That's the constant error I was getting when it was damaged.  When I replaced the error went away.  

How did you format your USB drive and did you have to load any of the .img files mentioned in the FF examples?

mbirth, 2019-12-31:
@oldtrusty Sorry, no. Never fiddled with my Gotek and/or dumped the fw.

kkwannyc, 2020-04-07:
Hi - fellow Flashfloppy user here on the SFR1M44-DU26.  Mine was used with an IBM slimline floppy drive enclosure to allow an oldschool ThinkPad (560E) to be used with it.  Mine has a revised logic board (SFRC2DA) that looks to be almost the same as the SFRC2D.  

Just to confirm with the postings of @JoanMCD - you do have to remove/disconnect resistors RA, RB and RC for the device to get out of flash update mode.  Once it's removed a typical FAT32 formatted flash drive will work just fine.  

- the standard Gotek bottom pin-outs as seen on the Flashfloppy modifications guide will work with this model.  

- Note that the 26 pin FFC model works on 5v, and pins 1, 3 and 5 are power input rails (with the buttons/USB port on top and the 26 pin ribbon connector on the bottom, Pin 1 is on the left side of the connector).  If you need to run continuity checks between the 5v pin and the power, just test pin 1 on the CPU (see below) with the 5v pin on the pinout, and pins 1, 3 or 5 on the ribbon cable connector.  

Namely:
[Boot0] [3.3] [TX] [RX] [RST]
                                [5v] [GND]

The following pins on the output corresponds to the following pinout on the STM104's LQFP64 package:

5v - Pin1 (VBAT)
Boot - Pin 60 (Boot0)
TX - Pin 42 (J?, next to J4, PA09)
RX - Pin 43 (J4, PA10)

Note that on the SFRC2D/A the LCD connector is the following from left to right:

[SDA] [SCL] [GND] [3.3V]  

SCL - PIn 29
SDA - Pin 30

On my 2 row LCD the output is rather faint.  I might have to do something about it. 

To program the SFRC2D/A with a USB-TTL serial adapter (preferably one that can do both 3.3/5v with the flick of a jumper), put a jumper between Boot0 and 3.3v, TX and TX to the correct pins, run VCC (5v please) to the 5v pin, and ground to the GND pin, and follow the usual instructions.

Note that there are existing test points on the board that can be used to verify that the soldering has been done correctly.   See the attached photo for an idea of where they are located.

[image]

And yes, Flashfloppy is so much better than the default firmware.  Try loading the 1.72MB OS/2 Warp 3 install disk set using the default firmware...it doesn't work.

ledge64, 2020-12-23:
Hello. I am currently having some trouble with programming the SFRC2DA with the USB-TTL serial adapter. I followed the instructions given by @kkwannyc 

> To program the SFRC2D/A with a USB-TTL serial adapter (preferably one that can do both 3.3/5v with the flick of a jumper), put a jumper between Boot0 and 3.3v, TX and TX to the correct pins, run VCC to the 5v pin (below J4), and ground to the GND pin (below RST), and follow the usual instructions.

[image]

I removed the three resistors (RB, RC, and RA), then I followed the wiring diagram. I attached a picture of how I wired the SFRC2DA to the USB-TTL serial adapter. I then proceeded to follow the video from Kris Cochrane that was linked in the wiki page. After opening the ST Flasher program I got the following error attached. 

[image]

I was wondering what else I could be missing. 

Thanks

dchauran, 2021-05-12:
Just to add a success here:

Tektronix TDS640A
Board rev SFRC2DA
0.91" oled
Removed RA, RB, RC, left the eeprom.
Attached a microswitch in place of RA, as the pins correspond to JA and I wanted that third button, mounted by soldering it to the side of an existing button.
My board came with JD and JE installed, I had to remove those.
Shugart mode, not IBM as I have seen on other scopes.

Initial programming was performed in DFU/USB mode (jump 3.3v/boot and tx/5v, connect to PC and use the ) since I couldn't find my USB to serial adapter, worked perfectly. 

The drive is a bit narrower than the original drive, so there is a small gap in the front panel.

The drive seems slow, but i have not had a chance to play with the skew and interleave yet to see if that helps.

kkwannyc, 2021-08-21:
> Hello. I am currently having some trouble with programming the SFRC2DA with the USB-TTL serial adapter. I followed the instructions given by @kkwannyc
>
>
> [image]
>
> I removed the three resistors (RB, [...]

Check continuity with the test points/ pins I’ve given and see if there are some cold solders/bad contacts somewhere (it happens sometimes).  Also, I think you need to unlock the device for write before it will work, but if the commands are downright being ignored it might be a communications issue (hence…continuity checks)

pdieng, 2021-09-10:
> 
> 
> Just to add a success here:
> 
> Tektronix TDS640A
> Board rev SFRC2DA
> 0.91" oled
> Removed RA, RB, RC, left the eeprom.
> Attached a microswitch in place of RA, as the pins correspond to JA and I wanted that third button, mounted by soldering it to the side of an existing button.
> My board came with JD and JE installed, I had to remove those.
> Shugart mode, not IBM as I have seen on other scopes.
> 
> Initial programming was performed in DFU/USB mode (jump 3.3v/boot and tx/5v, connect to PC and use the ) since I couldn't find my USB to serial adapter, worked perfectly.
> 
> The drive is a bit narrower than the original drive, so there is a small gap in the front panel.
> 
> The drive seems slow, as it takes about 10 seconds to write 38KB, or around 30kbit/second, but I didn't think to time how long it took on an actual floppy drive.
> 
> [image]

I have a TDS640A and I am trying to do the exact same thing as you.  I have Flash Floppy firmware on my SFRC2DA and removed RA, RB and RC.  However when ever I try to plug in a thumb drive, it gives me error 34.  I have a freshly formatted thumbdrive with just FF.CFG and IMG.CFG.  When I plug it in it also creates the empty IMAGE_A.CFG file.

What do you have on your thumbdrive?  Do you have any jumpers installed?

It might be because I don't have a working thumbdrive setup yet but also when I press the "Hard Copy" button on the scope, it tells me that the "File System Media is Missing".

hara3d, 2021-11-29:
Thanks, I also success 
Lecroy LC574ALS (FDD was Citizen W1D)
Gotek SFR1M44-DU26 / SFRC2D.B(Artery AT32F415RBT7)
OLED(0.91)
FlashFloppy v3.29(Native mode)
S1 short

With only RA removed, lecroy didn't seem to recognize the replacement, even if I changed discs (.img) with left and right buttons.
With RA, RB, and RC removed, reloading directry can be selected after changed discs.
So if we use only one disk image, it seems good to remove only RA.

coredump4, 2022-01-30:
Thanks to all for this informative thread, it really helped me.
One tidbit I'll add is, if you're having trouble getting a Gotek (slim or regular) into DFU mode, try a different USB cable.  I had been trying to use a modern, good-quality USB3 A-A cable that did not work, but an older USB2 cable did.  HTH.

hisink, 2022-03-10:
Another success on TDS3032. Some road blocks, but eventually all solved. Here are the brief procedures.
1. connect the du26 to pc with boot0 pin short to the next hole for dfu load.
2. install Artery driver. (difficult to find. especially after they change domain name.) I use the version 1.5.47
3. flash the FF version 3.30 with Artery dfu software.
4. remove ra/rb/rc, install LED and remove JD,JE.
5. cut the plastic at the end of the gotek enclosure. the one I bought is about 3mm longer than the original floppy drive. so it can't fit into the chassis. After shortening it, then install it back to the oscilloscope
6. put the usb drive, format to fat32, and copy in an image. 
the everything just work.

pbernicchi, 2022-05-03:
@kkwannyc I've pored through your writeup and tried to glean from your pics here and on VOGONS what your jumper setup might be... e.g., you have JE/JD/S1 jumped in your picture above but I can't tell from the pics on VOGONS because the Gotek is closed up.

I'm trying like hell to get the same Artery DU26 working on my ThinkPad 560 AND 570. I'm getting a green LED during POST but "Operating System not found" on known, good images.

Would you mind attaching your FF.CFG as well so perhaps we can tag this for (specifically) early ThinkPads with the 26-pin external connector?

Wierzbowsky, 2022-07-18:
> Another success on TDS3032. Some road blocks, but eventually all solved. Here are the brief procedures.
>     1. connect the du26 to pc with boot0 pin short to the next hole for dfu load.
>     2. [...]

@hisink Thanks for the clear instructions. Is there a need to put back RA/RB/RC and any of the jumpers after reprogramming?

keirf, 2022-07-18:
They correspond to jumpers JA JB JC on normal Goteks. No need to reinstall those resistors.

Wierzbowsky, 2022-07-18:
> They correspond to jumpers JA JB JC on normal Goteks. No need to reinstall those resistors.

So, only S0 or S1 jumper should remain?

keirf, 2022-07-18:
I think so. If JC is actually correct you can get same effect with interface=ibmpc in FF.CFG. But I think shugart is probably correct.

Wierzbowsky, 2022-07-21:
I will test once I get my hands on the device.

Wierzbowsky, 2022-07-28:
Removed 3 solder jumpers and 2 physical jumpers and flashed with FlashFloppy. Works like a charm in my Dual-Slim setup. Many thanks for your excellent work, @keirf!!

[image]

Wierzbowsky, 2022-08-06:
I've created an article with links to the repositories.

Olas-Audio, 2022-10-06:
Hi everyone. A few years ago, I got an Agilent 54622A oscilloscope for about 100 USD, a real bargain and the truth is that it is an excellent measurement equipment, despite being just 100 MHz. I just want to share the methodology that I followed to be able to use a Gotek floppy emulator. Without having read this issue, I decided to buy the cheapest emulator of this brand (34 USD shipping included), it’s an SFR1M44-DU26 SFRC2D.B version. So, there is no need to buy an expensive type to perform this operation.

Step 1. Download FlashFloppy version 3.29 (I think 3.3 could also be used).
https://github.com/keirf/FlashFloppy/wiki/Downloads

Step 2. Download the file Artery_ISP_Programmer_V1.5.46 or later and install the driver “Artery_DFU_DriverInstall.exe”.
https://www.arterytek.com/en/product/AT32F415.jsp#Resource
https://www.arterytek.com/download/TOOL/Artery_ISP_Programmer_V2.0.06.zip

[Artery_ISP_Programmer_V1.5.46.zip](https://github.com/keirf/flashfloppy/files/9691279/Artery_ISP_Programmer_V1.5.46.zip)

Step 3. Prepare the PCB for flashing according to the follow images:

[image]
This version uses the Artery microcontroller.

[image]
SFRC2D.B version.

[image]
I used a USB type A cable on one end and on the other side I exposed the wires so I could solder them to the PCB. I performed this step since I didn't have a USB A-A type cable.

[image]
I soldered the wires in the shown sequence to the base of the USB connector on the PCB, the layout is as follows:
|GND-black|D+ Green|D- White|+V-Red|

[image]
I soldered a 5-pin header to put a jumper between Boot-3V pins. I only left the S1 jumper in place.

[image]
Install the driver.

[image]
Then run the app “ArteryISSProgrammer.exe”.

[image]
Select USB DFU as the port type. Then we proceed to connect USB cable.

[image]
Select language and click next

[image]
Click next

[image]
Click next

[image]
You must select Enable/disable protection, click yes from the dialog box and then click next

[image]
Wait until it’s done then click back

[image]
Click "Download to device" and then click the "add" button.

[image]
Look for the Hex file of the version you want to install, in my case 3.29.

[image]
Select "Global erase" and "verify after download" then click next

[image]

[image]

Finally click close. You can unplug the USB cable from the computer.

Now you can desolder the resistors RA, RB and RC and the USB cable in any case. The memory IC does not need to be preserved. I removed it so I could use it in some other project later on.

[image]

[image]

Don't forget to remove Jumper Boot-3.3V.

Finally, it is necessary to prepare you USB device. Give it a format FAT32 and then you need to copy the "1m44_fat12.img" file into the main directory of the USB that you just formatted.
https://github.com/andrewpono/TDS3000-Series-Gotek-FlashFloppy/blob/main/1m44_fat12.img

That’s it, your scope should be ready to save Images into your USB stick. The file storing time it’s slow but it’s definitely more convenient than to use a Floppy Disk.

Cheers and enjoy your measurement!

kkwannyc, 2023-01-09:
> @kkwannyc I've pored through your writeup and tried to glean from your pics here and on VOGONS what your jumper setup might be... e.g., you have JE/JD/S1 jumped in your picture above but I can't [...]

Hah, sorry - I really need to log into my personal Github more often.  JD/JE/S1 are indeed jumped on the Gotek.  I haven't actually used it in the past few months since the OLED display on my custom enclosure just kept dying, and I only use it on my ThinkPad 560E (I am not sure where the cable for my 240 went...)  

The Gotek (with an STM32) only have the following on its ff.cfg:

interface = ibmpc 
host = pc-dos 
pin02 = auto 
pin34 = auto 
write-protect = no 
max-cyl = 255 
side-select-glitch-filter = 0 
track-change = instant
write-drain = instant 
index-suppression = yes
head-settle-ms = 12 
motor-delay = ignore
chgrst = step
ejected-on-startup = yes
image-on-startup = last
display-probe-ms = 3000
autoselect-file-secs = 1
autoselect-folder-secs = 2
folder-sort = always
sort-priority = folders
nav-loop = yes
twobutton-action = zero
display-type =  oled-128x32
oled-font = 6x13
oled-contrast = 145

[image]

KingDaveRa, 2023-03-18:
Hi all,

Just wanted to add my tuppence worth.

I've got the SFRC20.B and have successfully re-flashed it and got it working with a Toshiba T1900C laptop - they are very fussy as they use a slightly odd Citizen floppy drive (which has an unobtainium drive belt) and uses RDY on pin 34. 

I did the following:

1. Added pin headers as normal. Flashed using the standard methods.
2. Removed RA resistor
3. Jumpers as default

Relevant config in the FF.CFG:

`interface = ibmpc
host = unspecified
pin02 = auto
pin34 = nc
`
And that was it, booted fine. I hope this helps somebody else!

csulaguy, 2023-07-31:
> Hi - fellow Flashfloppy user here with the SFR1M44-DU26. Mine was used with an IBM slimline floppy drive enclosure (formerly occupied by a TEAC FD05HF) to allow oldschool ThinkPads (560E) to work off disk images and whatnot. Mine has a revised logic board variant (SFRC2DA) that looks to be almost the same as the SFRC2D.
> 
> [image]
> 
> Just to confirm with the postings of @JoanMCD - you do have to remove/disconnect resistors RA and RC for the device to get out of flash update mode. Once it's removed the device will work just fine. Retaining the RB+EEPROM (used by the original firmware to...I don't even know what it does as this Gotek variant does not behave like normal Goteks, so formatting your drive into multiple segments of 1.44MB in size is not going to work)...is purely up to you. I disconnected RB but kept the EEPROM alone on mine.
> 
> Things to be aware of -
> 
> * The 26 pin FFC model runs on 5v, pins 1, 3 and 5 on the ribbon cable connector are power input rails (pin 1 is on the very left side of the FFC connector).  If you need to run continuity checks between the 5v pin and the power, test pin 1 on the CPU (see below) with the 5v pin, and pins 1, 3 or 5 on the ribbon cable connector.
> * the standard Gotek bottom pin-outs from the Flashfloppy modification guide will also work with this model...you would have to solder the pins on and check your work accordingly (see test point locations on the photo and described later)
> 
> [Boot0] [3.3] [TX] [RX] [RST] [5v] [GND]
> 
> The following pins on the output corresponds to the following pinout on the STM32F105 LQFP64 package, and with the colors noted on the diagram below.
> 
> * The LQFP64 package contains 16 pins per side, 4 sides, and pin 1 starts on the bottom row on the left and proceeds counter-clockwise around the chip until pin 64 on the left side of the CPU on the bottom.
> 
> 5v - Pin1 (VBAT) Red Boot0 - Pin 60 (Boot0), Purple TX - Pin 42 (J? next to J4, PA09), Green RX - Pin 43 (J4, PA10), Blue
> 
> There is an AMS1117 voltage regulator(SOT223 package) with 3 pins, denoted on the board as U4. Left is GND, middle is 3.3v, and right is 5v.
> 
> On the SFRC2D/A the display connector is the following from left to right:
> 
> [SDA] [SCL] [GND] [3.3V] SCL - PIn 29 (PB10) - Yellow SDA - Pin 30 (PB11) - Orange
> 
> Ground is denoted as Black, and 3.3v is denoted as Orange-Red.
> 
> To program the SFRC2D/A with a USB-TTL serial adapter (preferably one that can do both 3.3/5v with the flick of a jumper), put a jumper between Boot0 and 3.3v, TX and TX to the correct pins, run VCC to the 5v pin (below J4), and ground to the GND pin (below RST), and follow the usual instructions.
> 
> There are existing test points on the board that can be used to verify that the soldering has been done correctly. See the attached photo for an idea of where they are located. Test points for RX/TX is above the STM105, Boot0 is to the left of the STM32F105, SDA/SCL is to the right (SDA will require you to test continuity with pin 27 directly), and 5v is present in at least 4 locations + the right leg off U4. 3.3v is supposed to be on the rightmost pin of the display pinout and on the middle leg off U4. @keirf you might find this photo of the SFRC2DA useful:
> 
> [image]
> 
> Hope this bit is useful for those trying to make this device do useful things - the SFR1M44-GU26 have practically no open documentation available whatsoever besides "buy this overpriced Gotek variant" and a poorly translated instruction manual of questionable applicability. @mbirth The default firmware is utterly useless crap if you plan to work with non-standard disk images, like the 1.72MB XDF images used by IBM's OS/2. For 1.44MB floppies it's not that difficult to grab a disk image template and have the FDC write/save it to a blank disk image, save it and then rename it on the USB stick later.
> 
> @oldtrusty did your model come with the jumpers for the bottom headers? I noticed that on mine the JD/JE headers were jumped...not sure if this might impact your Oscilloscope reading the device.

I don't suppose you have a copy of your ff.cfg? I'm trying to run mine on a ThinkPad 755C. The drive will boot images, but sometimes has trouble switching images. Also, ThinkPad diags and software setup think there's no floppy installed (but swapping in the Teac FD-05HG it came with shows up), but I can still read it most of the time. Problematic areas are writing images with IBM's loaddskf or equivalent, rawrite2 for Linux, and reading the root image and beyond on Linux. I was, however, able to install OS/2 2.1 from start to finish.

pablogventura, 2023-12-02:
I followed these instructions along with the standard FlashFloppy installation guide and it worked perfectly on a variant SFRC2DA board. Even the OLED screen remained functional. It was to restore a laptop from the 90's, an Epson Action Note 4SLC/33. Thank you!

bodong001, 2024-09-27:
Hello everyone, I really need your help. Regarding this floppy drive simulator, can you provide photos of the back of the PCB to determine the direction of certain pins of the 26 pin connector. Thank you, may you all be happy in the long run!

Wierzbowsky, 2024-09-27:
@bodong001 Will this help? 
[image]

bodong001, 2024-09-28:
Thank you Wierzbowsky ! My problem has been solved. It was caused by me forgetting to close the SO jumper. In addition, according to your prompt, I modified pin 2 of pin 34. Now the oscilloscope is working very happily! Thank you again!

liveboxandy, 2024-10-06:
From pictures and the use of a meter I have created what should be a schematic for the above Artery version SFRC2D.B. Should be the same as the STM version as a couple of smoothing caps make no sense on the Artery but do on the STM105.

[Gotek.pdf](https://github.com/user-attachments/files/17269278/Gotek.pdf)
I have also created a Kicad project for the above at https://github.com/liveboxandy/Gotek-SFRC2DB. Hope someone finds this useful.

keirf, 2024-10-07:
@liveboxandy Thanks, I have linked to your repository from the FlashFloppy wiki.

bodong001, 2025-11-20:
> hello guys, I am new user.Recently I step by step on the web and form for setup.now I can access the file on floppy emulator. But some problem I want to ask ,When I use tektorix 3000 series,I click the save ch1 waveform to file on oscillator , and it jump a icon of timer... and wait long time about 30min , and still no update.So I put poweroff. and a few minute and power on I can see a file on my usb disk. It it normal? I watch other people operator usb access on youtube , link is ....https://youtu.be/pQ0aSoYaUhY?si=IDZnGWkU6GEWDmHe it seems not long time for save .In my case ,my machine look like no trigger to oscillator to close written by floppy emulator.... andbody can help?

It may be a problem with your method. To save the waveform, I remember to press the print button on the left side of the screen. In addition, even the original floppy drive will not format the floppy disk properly in the oscilloscope system.

Cnytntg927, 2025-11-20:
> Follow the instructions in https://github.com/keirf/flashfloppy/wiki/Firmware-Update using the files in the /alt/logfile folder. That should install the correct firmare.

So strange when I click update flash and it blink look like finish and reboot ;but the reboot jump ff updates flash in loop there is nothing

[image]

Cnytntg927, 2025-11-21:
> I have just installed FF3.44 alt/logfile and copied setup and a waveform to the stick with no issues.
> 
> I am certain that you need a cable WITH the connections reversed as per https://github.com/andrewpono/TDS3000-Series-Gotek-FlashFloppy under HARDWARE. If you have a straight cable it will not work. Your TDS3054 might have a cable with the connections swapped, if not you NEED to change the cable. Ensure that the programming adapter has been removed from the Gotek. The youtube video you refer to is still using the original Gotek firmware, not FlashFloppy so the method of copying too and from is totally different.
> 
> Don't know how to test the Gotek without using the scope unless you have another system that uses 26Pin FFC.

I’m sure the cable is the same as you mentioned. Also, did you see the video I posted above? It actually works, but very slowly. That’s why I’m confused.If it were a build or firmware problem, I would expect no file access at all, but in the video you can see that it does access the file… just extremely slowly. It’s very strange.I’m even considering buying a new emulator because I really don’t know what’s causing this.

liveboxandy, 2025-11-21:
I can't seem to locate the video you posted of your Gotek saving. Can you point me to it?
Don't understand why your one doesn't work. 
Can you try installing the alt/logfile version and then post the created log file after it has booted and mounted an image file.

Cnytntg927, 2025-11-21:
> [@Cnytntg927](https://github.com/Cnytntg927) If your GOTEK is converted from 26 pin to 34 pin, please ensure that the 8nd pin of 26 pin is connected to the 34th pin of 34 pin. Alternatively, you can achieve this goal by configuring the CFG file.

This fourm isnt disscu the version for 26 pin emulator ??? Tektronix 3000 series only has 26 pin interface is didn't has other interface. And I sure that i am connect but you can see our disscu above post. Someone says no ff.cfg file need to set,it only insert can used and very quickly. The result isnt same as I know which you can watch above video I have post the case above post.

Dietrich-L, 2025-12-09:
Well, the data Transfer rate is the same as with a Floppy. Just the seek times are approx. Nil.
So yes, it feels more Like a Floppy than a modern SD
Dietrich