keirf/flashfloppy #157: Flash256KB and KANJI DISPLAY

SuperTurboZ, 2018-08-21:
Hello, keirf.
First, thank you for a great work.

I am developing GOTEK-FlashFloppy's KANJI filename display.
https://twitter.com/SuperturboZ/status/1028665995303444481
It seems to work already well, but its size does not fit the 128KB MCU because the OEM and UniCode conversion tables are too big.
It was necessary to change the size of FlashMemory to 256 KB, but it can not be changed with switch of makefile.

As a first step, I did make changes to switch between 128 KB and 256 KB.
As a next step, I will modidy to support kanji display. This is not necessary for people other than Japanese, so it may not be taken into the master.

Are you interested in my work?

I started learning Git and GitHubn yesterday, so I am a beginner.
I am also not good at English.
Sorry if I came off as rude.

keirf, 2018-08-24:
This is great! Feel free to share your work on Github or elsewhere and I can advise on whether it is suitable for inclusion into the main tree, but if not you have my blessing to continue it as a forked project.

SuperTurboZ, 2018-08-24:
Tank you.

I sent my first pull request abut Flash 256 KB support.
Please let me know if I have problems with my procedure.

I will continue to change Kanji support.
It is completed, but it takes a little time to learn the Git branch.

keirf, 2018-08-24:
Yes, the principle is fine. I wouldn't take that patch as is via pull request of course as it would break the main build.

I think more broadly I would like to define an Options file that can include high-level things like version=v0.9.27a and language=kanji, and then cook that via a Python script into lower-level things like Flash mem size... it's somewhat bikeshedding though.

closed 2018-09-12 by keirf